google-deepmind/mujoco #798: Offscreen rendering while using python interactive viewer

vikashplus, 2023-04-01:
How should one go about getting offscreen rendering to work while using the mjpython which is required by the interactive python viewer.  Here is a small snippet of code to explain my usecase 
``` python
# mujoco_viewr_test.py
    # load model 
    sim_path = 'humanoid.xml'
    model = mujoco.MjModel.from_xml_path(sim_path)
    data = mujoco.MjData(model)

    # start a passive viewer
    viwer = mujoco.viewer.launch_passive(model, data)

    # offscreen rendering
    renderer = mujoco.Renderer(model)
    renderer.update_scene(data)
    rgb_arr = renderer.render()
    print(rgb_arr.shape)
```

1. `python mujoco_viewr_test.py` throws `RuntimeError: launch_passive requires that the Python script be run under mjpython`
2. `mjpython mujoco_viewr_test.py` throws `mjpython[29249:31035860] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'`

drg3nz0, 2023-04-07:
i have the same error, is it with M2 Apple ?

vikashplus, 2023-04-07:
No, I am on mac but not on M2.

saran-t, 2023-04-07:
Can you please try https://github.com/saran-t/mujoco/commit/e570cb57e8530d1d38990af9d572141acae4b54e and see if this resolves the issue?

peterdavidfagan, 2024-01-02:
+1 just encountered the same error. @saran-t is there a chance the fix you give will be merged in the next release? I haven't tested it yet but plan to, eager to understand if it will be included in future releases.

commit 6d63e046 (2023-04-10): Add low-level CGL context management for macOS.

This allows rendering from non-main threads as the context is not tied to a Cocoa window.

Fixes #742
Fixes #798

PiperOrigin-RevId: 523144359
Change-Id: I98432debdd034814c313eda94e38c248c3c1f76c