google-deepmind/mujoco #799: How to fix ERROR: eigenvalue of mesh inertia too small for small mesh objects (Issue #759)

quangminh2702, 2023-04-02:
Hi,

I'm using MuJoCo for simulation of an assembly process. I decomposed an object into a union of few hundreds convex geoms using V-HACD and then using obj2mjcf to generate a XML file for that object.

When dropping that XML model into MJC 2.10, there exists an `error: eigenvalue of mesh inertia too small`

I think there is no issue with the mesh file. If I scale the mesh file up using `scale="1000 1000 1000"`, then it becomes big enough to avoid the small eigenvalue issue of mesh inertia. The up-scaled mesh file can then be rendered and seen as in Mujoco210. (I also checked the original mesh file using Blender to make sure that no face has inverted orientation)

However, we would like to work with the original size of the mesh file. Here are the [XML file](https://drive.google.com/file/d/1Vb123lKsztNhWLeIY70mKi7ER6IEDGUO/view?usp=sharing) and the [mesh file](https://drive.google.com/file/d/1qLcBpxa_dsPXK6IJuEVInJoS2Alz4T8s/view) that I am working around. 

Could you please give me some advices on how to fix the error?

I followed instructions from @yuvaltassa and @quagla to include both flags (`boundmass` and `boundinertia` in `compiler` and `shellinertia` in `geom`), but still there is an error `XML Error: Schema violation: unrecognized attribute: 'shellinertia'`.  If I removed `shellinertia="true"`, then **error: eigenvalue of mesh inertia too small** persists.

Thank you so much for your help!

_Originally posted by @quangminh2702 in https://github.com/deepmind/mujoco/issues/759#issuecomment-1486531165_

quagla, 2023-04-03:
Is there anything that prevents you from using the latest MuJoCo version?

quangminh2702, 2023-04-04:
Hi @quagla . Thanks for your suggestion. I updated to Mujoco 2.3.2 and the issue with shellinertia no longer exists.
However, I have another concern. By setting `shellinertia="true"`, we are assuming that all the mass is concentrated on the boundary, which is not physically true, isn't it?

commit 94c4f729 (2023-04-18): Specify is volume is too small or negative in mesh error messages. Fixes #799.

PiperOrigin-RevId: 525151156
Change-Id: Ia0e9d8b900b8c372e9229da934a4eb7aeb1a3b7a
closed 2023-04-18 by copybara-service[bot]