google-deepmind/mujoco #803: Using tendons to drive joints

thawke, 2023-04-06:
Hello, I'm trying to set up a cable-driven manipulator and have some questions I've not been able to resolve using the docs or by playing around with the minimal example of a hanging mass below. At the moment I'm using Python to get to grips with everything.

## Questions

1. Is it possible (and appropriate) to directly manipulate the position of the origin end of a tendon to simulate some position actuator using `data.site_xpos`? I haven't been able to make it have an effect, as seen in my example. I don't want to use an actuator directly on the tendon because:
    1. It will change the length of the tendon and I don't want this to affect the stiffness.
    1. I don't want any steady state error and will eventually put my own position controller around the origin position.

1. Related to 1.i.: Is the tendon stiffness coefficient *specific* stiffness - i.e. is it per unit length, with true linear stiffness calculated using `data.ten_length` at runtime?
1. Is it currently possible to have a nonlinear tendon stiffness? Ideally, I'd like a tendon that has zero stiffness in the compressive direction and a piecewise linear stiffness in tension.

## Playground
### `masstendon.xml`
```xml
<mujoco>
    <default>
        <site type="sphere" rgba="1 0 0 1" size="0.005"/>
        <tendon rgba="0 1 0 1" stiffness="1000" damping="100"/>
    </default>
    
    <visual>
        <headlight diffuse=".7 .7 .7"/>
    </visual>

    <worldbody>
        <site name="origin0" pos="0 0 0"/>
        <body name="mass" pos="0 0 -0.3">
            <freejoint/>
            <geom name="massgeom" type="box" size="0.05 0.05 0.05" pos="0 0 0" mass="1"/>
            <site name="insertion0" pos="0 0 0.05"/>
        </body>
    </worldbody>

    <tendon>
        <spatial name="rope0">
            <site site="origin0"/>
            <site site="insertion0"/>
        </spatial>
    </tendon>

    <sensor>
        <force name="tension0" site="origin0"/>
    </sensor>

</mujoco>
```
### `masstendon.py`
```python
import mujoco as mj
import mujoco_viewer
import numpy as np
import matplotlib.pyplot as plt

model = mj.MjModel.from_xml_path("masstendon.xml")
data = mj.MjData(model)
viewer = mujoco_viewer.MujocoViewer(model, data, hide_menus=True)

def main(model, data):
    forcetrace = []
    postrace = []

    for _ in range(1000):
        viewer.render()
        
        if _>500:
            data.site_xpos[0]=np.array([0, 0, 0.1]) # move tendon origin up a bit
        else:
            pass
    
        mj.mj_step(model, data)
        forcetrace.append(data.sensordata[2])
        postrace.append(data.qpos[2])
        
    viewer.close()
    
    return forcetrace, postrace

forces, posns = np.array(main(model, data))
plt.plot(forces)
plt.figure()
plt.plot(posns)
plt.show()
```

Any help welcome!

Thanks

yuvaltassa, 2023-04-06:
1. Did you read #520?
2. Stiffness is absolute, not specific.
3. Your other questions are unclear. Can you describe what you are trying to achieve, rather than the problem you think you've encountered?

thawke, 2023-04-06:
Thanks @yuvaltassa.

> Did you read https://github.com/deepmind/mujoco/issues/520?

Sorry, hadn't seen this issue. Very relevant and `springrange` solves the directional problem - thanks. I am interested in higher order spring stiffnesses. I'd rather base them on measured data (hence the piece-wise linear request), but I could fit a polynomial if needed.

> Stiffness is absolute, not specific.

Understood. I'll implement my own specific stiffness.

> Your other questions are unclear. Can you describe what you are trying to achieve, rather than the problem you think you've encountered?

Essentially, in a given timestep, I want to manually set the position of the end of a fixed length (but stretchy) tendon whose other end is connected to a body. I don't think the actuator mechanism can give me what I want since the output is always a force. Is this possible and how would I go about it?

nimrod-gileadi, 2023-04-06:
I think what you want can be achieved if you place the site inside a [mocap body](https://mujoco.readthedocs.io/en/stable/modeling.html#mocap-bodies), and move the mocap body (by setting values of mocap_pos), rather than directly modifying site_xpos.

yuvaltassa, 2023-04-06:
@nimrod-gileadi the OP still hasn't explained what it is that they are ultimately trying to model but IIUC they need the site in proper moving body, not a mocap body.

Both of these are possible (and easy), I'll make an example later today. A better explanation of the OP's goals would still be helpful.

thawke, 2023-04-06:
> I think what you want can be achieved if you place the site inside a [mocap body](https://mujoco.readthedocs.io/en/stable/modeling.html#mocap-bodies), and move the mocap body (by setting values of mocap_pos), rather than directly modifying site_xpos.

Thanks, @nimrod-gileadi, this looks promising. I'll test it tomorrow.

@yuvaltassa, sorry if I'm being unclear; my problem is conceptual rather than specific to one model. In general, I want to apply forces to bodies via attached tendons. I want to manipulate the positions of the free ends of the tendons. I don't care whether the free ends are attached to mocap bodies or regular bodies (or no bodies!), just that I can directly control their position for now. 

The end-goal, as shown below, is to model some arbitrarily complex serial manipulator, whose joints are actuated via cables that are routed along the manipulator to one location, where they are pulled using dedicated position servos (but I don't want to model these now).

[image]

yuvaltassa, 2023-04-06:
Thank you, this is helpful.

In this case you almost certainly don't want mocap bodies, as these are somewhat unphysical, having no velocities. What you want are bodies on linear joints with high armature/damping and high-kp position actuators (note "high" is relative, you'll need to tune this). Put a site on these bodies and attach a tendon to it.

More generally, I'd strongly recommend designing the models in simulate rather than Python. The fast reloading and interactivity make a world of difference.

yuvaltassa, 2023-04-15:
The model below looks like this. Hope this helps.

[image]

<details>
  <summary>slider model</summary>
  
```XML
<mujoco>
  <compiler autolimits="true"/>

  <default>
    <site type="sphere" rgba="1 0 0 1" size="0.005"/>
    <tendon rgba="0 1 0 1"/>
  </default>
  
  <visual>
    <headlight diffuse=".7 .7 .7"/>
  </visual>

  <worldbody>
    <body name="arm">
      <joint name="arm" axis="0 1 0" damping=".04"/>
      <geom name="shoulder" type="cylinder" size=".05 .01" zaxis="0 1 0"/>
      <geom name="arm" type="box" size=".01" fromto="0 0 0 -.3 0 0"/>
      <geom name="crossbar" type="box" size=".01" fromto="-.1 0 -.05 -.1 0 .05"/>
      <site name="arm" pos="-.1 0 .05"/>
    </body>
    <site name="right" pos=".1 0 0"/>
    <body name="slider" pos=".05 0 -.2">
      <joint name="slider" type="slide" damping="1"/>
      <geom name="slider" type="box" size=".01 .01 .01"/>
      <site name="slider" pos="0 0 .01"/>
    </body>
  </worldbody>

  <tendon>
    <spatial name="rope" range="0 .35">
      <site site="slider"/>
      <geom geom="shoulder" sidesite="right"/>
      <site site="arm"/>
    </spatial>
  </tendon>

  <actuator>
    <position name="slider" joint="slider" ctrlrange="-.05 .05" kp="100"/>
  </actuator>
</mujoco>
```

</details>

closed 2023-04-15 by yuvaltassa