google-deepmind/mujoco #804: Modify the tutorial code on Colab so that it can output images or videos in PyCharm on Windows.

EyeEyeloving, 2023-04-06:
Hello everyone, I ran the code example in the **Simulation** section on [tutorial.ipynb](https://colab.research.google.com/github/deepmind/mujoco/blob/main/python/tutorial.ipynb) and it ran successfully. Now, I hope to run this code example on **PyCharm** on **Windows**, but there is no code output and there is the following output prompt:

```python
<IPython.core.display.HTML object>
```

This example code is as follows.

```python
xml = """
<mujoco>
  <worldbody>
    <light name="top" pos="0 0 1"/>
    <body name="box_and_sphere" euler="0 0 -30">
      <joint name="swing" type="hinge" axis="1 -1 0" pos="-.2 -.2 -.2"/>
      <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
      <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
    </body>
  </worldbody>
</mujoco>
"""
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)
renderer = mujoco.Renderer(model)

# enable joint visualization option:
scene_option = mujoco.MjvOption()
scene_option.flags[mujoco.mjtVisFlag.mjVIS_JOINT] = True

duration = 3.8  # (seconds)
framerate = 60  # (Hz)

frames = []
mujoco.mj_resetData(model, data)
while data.time < duration:
  mujoco.mj_step(model, data)
  if len(frames) < data.time * framerate:
    renderer.update_scene(data, scene_option=scene_option)
    pixels = renderer.render()
    frames.append(pixels)

# Simulate and display video.
media.show_video(frames, fps=framerate)
```

In the last step

```python
media.show_video(frames, fps=framerate)
```

It has never been output, let alone the example code before it as follow.

```python
xml = """
<mujoco>
  <worldbody>
    <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
    <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
  </worldbody>
</mujoco>
"""
# Make model and data
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

# Make renderer, render and show the pixels
renderer = mujoco.Renderer(model)
media.show_image(renderer.render())
```

I want to ask, how should I modify the code that outputs pictures or videos at the end so that the entire program can run successfully and output in PyCharm on Windows?

Thanks a lot!

kevinzakka, 2023-04-08:
I've never used PyCharm before but I assume there's some extension that runs the notebook inside the IDE as in visual code?

`mediapy` is basically outputting HTML under the hood and that gets displayed inside the Jupyter notebook or Colab cell. If that's not working for you, you can write the video to disk and then view it locally. See [here](https://github.com/google/mediapy#videos) for an example.

kevinzakka, 2023-04-08:
Closing for now, feel free to re-open if you're still stuck.

closed 2023-04-08 by kevinzakka

EyeEyeloving, 2023-04-09:
I found a successful and useful example code from the [issues#798](https://github.com/deepmind/mujoco/issues/798#issue-1650628921) shown as follows.

```python
# mujoco_viewr_test.py
    # load model 
    sim_path = 'humanoid.xml'
    model = mujoco.MjModel.from_xml_path(sim_path)
    data = mujoco.MjData(model)

    # start a passive viewer
    viwer = mujoco.viewer.launch_passive(model, data)

    # offscreen rendering
    renderer = mujoco.Renderer(model)
    renderer.update_scene(data)
    rgb_arr = renderer.render()
    print(rgb_arr.shape)
```

I understand that the order of updating the scene and simulating the physical properties in programming is crucial. I’m thrilled to have found a useful example that can illustrate this writing rule to me.

Now, my runnable code is as follows.

```python
import mujoco
from mujoco import viewer

xml = """
<mujoco>
    <option gravity="0 0 10"/>
      <worldbody>
        <light name="top" pos="0 0 1"/>
        <body name="box_and_sphere" euler="0 0 -30">
          <joint name="swing" type="hinge" axis="1 -1 0" pos="-.2 -.2 -.2"/>
          <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
          <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
        </body>
      </worldbody>
</mujoco>
"""

model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)
renderer = mujoco.Renderer(model)

viwer = viewer.launch_passive(model= model, data= data)

# enable joint visualization option:
scene_option = mujoco.MjvOption()
scene_option.flags[mujoco.mjtVisFlag.mjVIS_JOINT] = True

duration = 3.8  # (seconds)
framerate = 10  # (Hz)

# Simulate and display video.
while data.time < duration:
    mujoco.mj_step(model, data)
    if data.time * framerate <= duration * framerate:
        renderer.update_scene(data, scene_option=scene_option)
        pixels = renderer.render()
        print(pixels.shape)
```

Thanks for your replying.

EyeEyeloving, 2023-04-09:
> I've never used PyCharm before but I assume there's some extension that runs the notebook inside the IDE as in visual code?
>
> `mediapy` is basically outputting HTML under the hood and that gets [...]

Thanks for your replying.

I’m grateful that the answer " `mediapy` is basically outputting HTML under the hood and that gets displayed inside the Jupyter notebook or Colab cell" is useful to me.