google-deepmind/mujoco #807: IOException, Error loading the XML, unsigned integer with an option suffix{K, M, G, T, P, E}

lzs0603, 2023-04-09:
Hi,

I noticed that for mujoco 2.3.2 and 2.3.3, while simulating the humanoid.xml using the Unity plugin with an intel chip Mac, the below error appeared and system paused. I tried using unityeditor 2021.3.21, 2021.3.1 and 2020.3.46, received the same error. For mujoco 2.3.1 and versions earlier, it seems to work just fine with not problem.

Here is a screenshot for the error:
[image]

erez-tom, 2023-04-19:
Is this a fresh import of an MJCF, or a saved scene?
If it's a fresh import, what's the value in the XML of `size`->`memory` (if at all)?
If it's a saved scene, do you have a `MjGlobalSettings` component?  What's the value in the field `Global Sizes`->`Memory`?

gpzhaome, 2023-07-06:
> Is this a fresh import of an MJCF, or a saved scene? If it's a fresh import, what's the value in the XML of `size`->`memory` (if at all)? If it's a saved scene, do you have a `MjGlobalSettings` component? What's the value in the field `Global Sizes`->`Memory`?

I had the same issue. I'm using a fresh import of MJCF humanoid.xml model. The memory is not set in the xml file. 

I fixed the issue by setting the `Global Sizes`->`Memory` to `2M` in the `MjGlobalSettings`.

News0412, 2023-07-12:
> I had the same issue. I'm using a fresh import of MJCF humanoid.xml model. The memory is not set in the xml file.
>
> I fixed the issue by setting the `Global Sizes`->`Memory` to `2M` in the [...]

I met the same issue after importing the example model (humanoid.xml). And I tried to set the `Global Sizes`->`Memory` to `-1`(the default setting) in the `Mj Global Settings` of `Global Settings`, it run well. So I think this might be a problem of parameter transmission between the mujoco and the unity.

Balint-H, 2024-02-01:
This issue has now been solved, and can be closed @erez-tom

closed 2024-08-30 by kevinzakka