google-deepmind/mujoco #813: How do I move objects at runtime? [C++]

marzRX, 2023-04-12:
Hi, I'm individual developer. I would like to make boxes fall one after another from the air and fall to the floor and pile up.

I think MuJoCo does not seem to be able to create objects dynamically, so I thought I would prepare boxes outside of the camera's view and programmatically move them around at runtime.
I wanted to prepare a box outside the camera's view and move it programmatically at runtime. However, the box does not move as intended and the camera only shows the floor all the time.

So, to simplify the problem, I created a very simple program. It is an xml file with one box at (0.5,0.0,0.1), I wrote a process in the main loop to move the box. We expect it to move to (0.0, 0.0, 0.1).

However, even this simple program does not work as intended. The box stays at (0.5,0.0,0.1) the whole time. I do not know what is wrong. Thank you in advance for your help.

The code is as follows. (main loop)

```
    mjtNum simstart = data->time;

    while (data->time - simstart < 1.0/60.0) {
      mj_step(model, data);

      double sim_time = data->time;
      if (sim_time - last_object_drop_time >= object_drop_interval) {

	if (object_id != -1) {
	  box_pos = data->xpos + 3 * object_id;
	  printf("[0] Box position: x=%.2f, y=%.2f, z=%.2f\n", box_pos[0], box_pos[1], box_pos[2]);

          // * NEW POSITION
	  mjtNum new_object_pos[3] = {0.0, 0.0, 0.1};
	  //	  mju_copy(&data->xpos[object_id * 3], new_object_pos, 3);
	  mju_copy(&data->qpos[object_id * 7 + 4], new_object_pos, 3);
	  mj_forward(model, data);

	  box_pos = data->xpos + 3 * object_id;
	  printf("[1] Box position: x=%.2f, y=%.2f, z=%.2f\n", box_pos[0], box_pos[1], box_pos[2]);
	} else {
	  printf("object_id = %d\n", object_id);
	}
	last_object_drop_time = sim_time;
	current_object_idx = (current_object_idx + 1) % 10;
      }
    }
```

The standard output is as follows.
```
[0] Box position: x=0.50, y=0.00, z=0.10
[1] Box position: x=0.50, y=0.00, z=0.10
```

yuvaltassa, 2023-04-12:
`xpos` is a function of `qpos` and gets overwritten every time you call mj_forward. 

This is explained in the documentation including [here](https://mujoco.readthedocs.io/en/latest/computation.html#the-state).

I strongly recommend working through our Python tutorial even though you are a C++ user, as it teaches some basic MuJoCo concepts.

closed 2023-04-12 by yuvaltassa

marzRX, 2023-04-14:
Thank you. I was able to solve the problem.
I used the following URL as a reference.

https://mujoco.readthedocs.io/en/stable/overview.html#floating-objects