google-deepmind/mujoco #816: Tendon invweight0 not correctly computed?

kracon7, 2023-04-12:
Hi,
I'm trying to add some extra constraints to the cloth object so I attempted to replicate the composite object wrapper with basic components such as joints, tendons, and equality constraints.

I made a 5x5 grid kinematic tree and closed the loop with tendons with equality constraints. All the components use default physics parameters.

[image]

However, the simulation is unstable. I used Python binding to examine the compiled model. Looks like the only difference is the tendon_invweight0. 
The model created from the given example in this repo has nonzero tendon_invweight0 while the one I replicated has zero.

I'm using:
  - Operating system: Ubuntu 20.04
  - MuJoCo version: 2.3.3

Here is the XML file to reproduce this issue. You can place it in the model/composite directory. Thank you!

```XML
<mujoco model="Cloth">
  <include file="scene.xml"/>

  <option timestep="0.002" solver="CG" tolerance="1e-6"/>

  <size memory="10M"/>

  <worldbody>
    
    <body name="Cloth_5_5" pos="0 0 0.1">
      <joint type="free"/>
      <geom name="c_0_0" type="sphere" size="0.015" rgba=".8 .2 .1 1"/>
      <site name="s_0_0" type="sphere" size="0.015" rgba="0 1 1 0.2"/>

      <body name="c_0_1" pos="0 0.05 0">
        <joint type='hinge' axis='1 0 0' />
        <joint type='hinge' axis='0 1 0' />
        <geom name="c_0_1" type="sphere" size="0.015" rgba=".8 .2 .1 1"/>
        <site name="s_0_1" type="sphere" size="0.015" rgba="0 1 1 0.2"/>

        <body name="c_0_2" pos="0 0.05 0">
          <joint type='hinge' axis='1 0 0' />
          <joint type='hinge' axis='0 1 0' />
          <geom name="c_0_2" type="sphere" size="0.015" rgba=".8 .2 .1 1"/>
          <site name="s_0_2" type="sphere" size="0.015" rgba="0 1 1 0.2"/>
        
          <body name="c_0_3" pos="0 0.05 0">
            <joint type='hinge' axis='1 0 0' />
            <joint type='hinge' axis='0 1 0' />
            <geom name="c_0_3" type="sphere" size="0.015" rgba=".8 .2 .1 1"/>
            <site name="s_0_3" type="sphere" size="0.015" rgba="0 1 1 0.2"/>
        
            <body name="c_0_4" pos="0 0.05 0">
              <joint type='hinge' axis='1 0 0' />
              <joint type='hinge' axis='0 1 0' />
              <geom name="c_0_4" type="sphere" size="0.015" rgba=".8 .2 .1 1"/>
              <site name="s_0_4" type="sphere" size="0.015" rgba="0 1 1 0.2"/>
        
              <body name="c_1_4" pos="0.05 0 0">
                <joint type='hinge' axis='1 0 0' />
                <joint type='hinge' axis='0 1 0' />
                <geom name="c_1_4" type="sphere" size="0.015" rgba=".8 .2 .1 1"/>
                <site name="s_1_4" type="sphere" size="0.015" rgba="0 1 1 0.2"/>
        
                <body name="c_2_4" pos="0.05 0 0">
                  <joint type='hinge' axis='1 0 0' />
                  <joint type='hinge' axis='0 1 0' />
                  <geom name="c_2_4" type="sphere" size="0.015" rgba=".8 .2 .1 1"/>
                  <site name="s_2_4" type="sphere" size="0.015" rgba="0 1 1 0.2"/>
        
                  <body name="c_3_4" pos="0.05 0 0">
                    <joint type='hinge' axis='1 0 0' />
                    <joint type='hinge' axis='0 1 0' />
[... 183 lines cut ...]
    <spatial name="tendon_16" width="0.005" damping="0.001" rgba="0.5 0.5 0.5 0.3" >
      <site site="s_4_4" /><site site="s_4_3"/>
    </spatial>
	</tendon>

  <equality>
    <tendon name='tendon_1' tendon1='tendon_1'/>
    <tendon name='tendon_2' tendon1='tendon_2'/>
    <tendon name='tendon_3' tendon1='tendon_3'/>
    <tendon name='tendon_4' tendon1='tendon_4'/>
    <tendon name='tendon_5' tendon1='tendon_5'/>
    <tendon name='tendon_6' tendon1='tendon_6'/>
    <tendon name='tendon_7' tendon1='tendon_7'/>
    <tendon name='tendon_8' tendon1='tendon_8'/>
    <tendon name='tendon_9' tendon1='tendon_9'/>
    <tendon name='tendon_10' tendon1='tendon_10'/>
    <tendon name='tendon_11' tendon1='tendon_11'/>
    <tendon name='tendon_12' tendon1='tendon_12'/>
    <tendon name='tendon_13' tendon1='tendon_13'/>
    <tendon name='tendon_14' tendon1='tendon_14'/>
    <tendon name='tendon_15' tendon1='tendon_15'/>
    <tendon name='tendon_16' tendon1='tendon_16'/>
  </equality>
</mujoco>
```

kracon7, 2023-04-12:
I can also confirm that the stability issue in the simulation is caused by tendon_invweight0 being 0.
After loading the model from XML, if I manually set ```model.tendon_invweight0``` to nonzero value (I used 100), the simulation is stable.
I also checked the source code in ```src/user/user_composite.cc``` to see how the composite object wrapper works. Seems all the tendons and equality constraints on the tendons use the default parameters.

yuvaltassa, 2023-04-15:
`tendon_invweight0` is computed [here](https://github.com/deepmind/mujoco/blob/4204109a5a852d2338c07ea739e2af63983a2d88/src/engine/engine_setconst.c#L139). It's not immediately clear to me why your tendons have 0 value. Possibly a bug?

Are you able to debug this (or at least gain understanding) given the above pointer to the code? The first thing I would try is see if you get different values with different settings of [option-jacobian](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-jacobian), as dense and sparse models use different code paths.

yuvaltassa, 2023-12-15:
Closing due to lack of response.

closed 2023-12-15 by yuvaltassa