google-deepmind/mujoco #817: (Python) Unable to use media: Exception ignored in: <function GLContext.__del__ at 0x0000017F60ED4430>

mertalagozlu, 2023-04-13:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a (student) and I'm trying to use MuJoCo for academic purposes.

I'm looking for some help with rendering simple simulation results with Python PyPI.

Here is the code which explains my question:

After installing mujoco and mediapy using pip, I am having type error as 'NoneType object is not callable'. Full error text is below.

xml = """
<mujoco>
  <worldbody>
    <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
    <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
  </worldbody>
</mujoco>
"""
model = mujoco.MjModel.from_xml_string(xml)
renderer = mujoco.Renderer(model, 480, 480)
data = mujoco.MjData(model)
mujoco.mj_forward(model, data)
renderer.update_scene(data, "fixed")
media.show_image(renderer.render())

**Error:**

Exception ignored in: <function GLContext.__del__ at 0x0000017F60ED4430>
Traceback (most recent call last):
  File "C:\Users\User\Desktop\Mujoco Converter\mujoco\env\lib\site-packages\mujoco\glfw\__init__.py", line 41, in __del__
  File "C:\Users\User\Desktop\Mujoco Converter\mujoco\env\lib\site-packages\mujoco\glfw\__init__.py", line 35, in free
TypeError: 'NoneType' object is not callable

Thanks for the help, I am fairy new to Mujoco PyPI and its dependencies. Is it possible to render simulation results it with any other library rather than **mediapy**?

saran-t, 2023-04-13:
This is an ignored error and shouldn't have stopped the renderer from working. Are you actually not getting images out of it?

mertalagozlu, 2023-04-13:
Nope, nothing is shown. I changed the __init__.py as removing '()' from the free method, but still no images. Is it possible that PyCharm doesn't show the images, yet I need to use Jupyter? Not entirely sure, they're related though.

saran-t, 2023-04-14:
Your original question didn't mention PyCharm. Please run the script in a standalone Python setup and see if you can get the image there.

closed 2023-04-30 by kevinzakka

manavkulshrestha, 2023-07-31:
Are there any updates in this? Running into the same issue in a standalone Python setup

caro33, 2023-09-08:
I'm also facing the same problem... Please let me know if somebody solved problem

hub2nature, 2024-01-23:
I also faced the same issue 
Let me know if there is any solution.

Exception ignored in: <function GLContext.__del__ at 0x7d56d6604820>
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/mujoco/glfw/__init__.py", line 41, in __del__
  File "/usr/local/lib/python3.10/dist-packages/mujoco/glfw/__init__.py", line 35, in free
AttributeError: 'NoneType' object has no attribute 'get_current_context'

kevinzakka, 2024-01-23:
This is fixed in 1c7809e54261009698b5baab569faf700c8f8dad and will be available in the next MuJoCo release.

dylandronnier, 2024-02-13:
I'm facing the same issue with Mujoco v3.1.2.

YitzhakSp, 2024-03-05:
> This is fixed in [1c7809e](https://github.com/google-deepmind/mujoco/commit/1c7809e54261009698b5baab569faf700c8f8dad) and will be available in the next MuJoCo release.

I'm facing the same issue although I use mujoco 3.1.2 that contains updated version render.py. please help.

kevinzakka, 2024-03-06:
@YitzhakSp Can you give us more information, especially a MWE and computer specs?

gpans, 2024-05-06:
Hi,

I am also experiencing the same issue, with nothing opening with the renderer. My code is as follows:

import mujoco
import mediapy as media

# import model parameters
xmlfile = "xml_files/terrain.xml"

# make model and data (for kinematics)
model = mujoco.MjModel.from_xml_path(xmlfile)
data = mujoco.MjData(model)
mujoco.mj_kinematics(model, data)

# Make renderer, render and show the pixels
renderer = mujoco.Renderer(model)
media.show_image(renderer.render())

Error: 
.venv) (base) PS C:\Users\User\Documents\Libraries\mujoco> python .\terrain.py
<IPython.core.display.HTML object>
Exception ignored in: <function Renderer.__del__ at 0x000001EEBF1880D0>
Traceback (most recent call last):
  File "C:\Users\User\Documents\Libraries\mujoco\.venv\lib\site-packages\mujoco\renderer.py", line 330, in __del__
  File "C:\Users\User\Documents\Libraries\mujoco\.venv\lib\site-packages\mujoco\renderer.py", line 316, in close
  File "C:\Users\User\Documents\Libraries\mujoco\.venv\lib\site-packages\mujoco\glfw\__init__.py", line 35, in free
TypeError: 'NoneType' object is not callable
Exception ignored in: <function GLContext.__del__ at 0x000001EEB03B5DC0>
Traceback (most recent call last):
  File "C:\Users\User\Documents\Libraries\mujoco\.venv\lib\site-packages\mujoco\glfw\__init__.py", line 41, in __del__
  File "C:\Users\User\Documents\Libraries\mujoco\.venv\lib\site-packages\mujoco\glfw\__init__.py", line 35, in free
TypeError: 'NoneType' object is not callable

System info:
Python: 3.8.0
Mujoco 3.1.4
Visual Studio Code: 1.89
Windows 11 Home

scakki, 2024-09-18:
Hello @gpans,
Could you let me know if you were able to solve this issue? I am also facing the same problem. 

System info:
Mujoco 3.2.3  
Python 3.9.19
Linux 22.04