google-deepmind/mujoco #818: Mouse perturbations not working in launch_passive viewer mode

rtkg, 2023-04-13:
Hi,

I'm using MuJoCo 2.3.3. via its Python bindings in Ubuntu. It seems that injecting mouse perturbations doesn't work with the new launch_passive mode of the viewer. The marker visualizations appear as usual, but to no observable effect. When launching the viewer regularly via `viewer.launch(model,data)`, everything works. 

Below I added a minimal example using the original hummanoid.xml model. 

<details>
  <summary>minimal XML</summary>

import mujoco
from mujoco import viewer
import time

xml = """
<mujoco model="Humanoid">
  <option timestep="0.005"/>

  <visual>
    <map force="0.1" zfar="30"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global offwidth="2560" offheight="1440" elevation="-20" azimuth="120"/>
  </visual>

  <statistic center="0 0 0.7"/>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1=".3 .5 .7" rgb2="0 0 0" width="32" height="512"/>
    <texture name="body" type="cube" builtin="flat" mark="cross" width="128" height="128" rgb1="0.8 0.6 0.4" rgb2="0.8 0.6 0.4" markrgb="1 1 1" random="0.01"/>
    <material name="body" texture="body" texuniform="true" rgba="0.8 0.6 .4 1"/>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1=".1 .2 .3" rgb2=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
  </asset>

  <default>
    <motor ctrlrange="-1 1" ctrllimited="true"/>
    <default class="body">

      
      <geom type="capsule" condim="1" friction=".7" solimp=".9 .99 .003" solref=".015 1" material="body"/>
      <default class="thigh">
        <geom size=".06"/>
      </default>
      <default class="shin">
        <geom fromto="0 0 0 0 0 -.3"  size=".049"/>
      </default>
      <default class="foot">
        <geom size=".027"/>
        <default class="foot1">
          <geom fromto="-.07 -.01 0 .14 -.03 0"/>
        </default>
        <default class="foot2">
          <geom fromto="-.07 .01 0 .14  .03 0"/>
        </default>
      </default>
      <default class="arm_upper">
        <geom size=".04"/>
      </default>
      <default class="arm_lower">
        <geom size=".031"/>
      </default>
      <default class="hand">
        <geom type="sphere" size=".04"/>
      </default>

      
      <joint type="hinge" damping=".2" stiffness="1" armature=".01" limited="true" solimplimit="0 .99 .01"/>
      <default class="joint_big">
        <joint damping="5" stiffness="10"/>
        <default class="hip_x">
          <joint range="-30 10"/>
[... 158 lines cut ...]
                            0 0.4 0
                            -0.25 -0.5 -2.5 -2.65 -0.8 0.56
                            -0.25 -0.5 -2.5 -2.65 -0.8 0.56
                            0 0 0 0 0 0"/>
    <key name="stand_on_left_leg" qpos="0 0 1.21948
                                        0.971588 -0.179973 0.135318 -0.0729076
                                        -0.0516 -0.202 0.23
                                        -0.24 -0.007 -0.34 -1.76 -0.466 -0.0415
                                        -0.08 -0.01 -0.37 -0.685 -0.35 -0.09
                                        0.109 -0.067 -0.7 -0.05 0.12 0.16"/>
  </keyframe>
</mujoco>
"""

model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

viewer.launch_passive(model, data)

while 1:
    # sleep is needed to avoid issue 783 (https://github.com/deepmind/mujoco/issues/783)
    time.sleep(0.001)
    mujoco.mj_step(model, data)

</details>

saran-t, 2023-04-13:
As its name suggests, `launch_passive` is _passive_. It won't modify your physics state. This is working as intended.

In the next few days I'll be releasing a viewer enhancement that lets you access and modify the passive viewer's visualisation state. You should then be able to apply the perturbations that are coming in from the viewer, but the point is that it's up to user code to actually apply these perturbations, not the viewer.

saran-t, 2023-04-13:
Having said that, perhaps it is actually reasonable for the passive viewer to also apply perturbations too. After all, it's already fiddling with various physics flags in response to various toggles (gravity, passive force, etc). What do people think?

rtkg, 2023-04-13:
Ah, I understand. Thanks for your quick reply and your efforts!

closed 2023-04-13 by rtkg

yuvaltassa, 2023-04-13:
@saran-t TBH, it should be super easy, right? 

The apply functions are already [in the `Simulate` object, just called from the physics loop](https://github.com/deepmind/mujoco/blob/018503a0ec024a5a7cd8bc98a825403eae1ce22f/simulate/main.cc#L362). So exposing them as Python viewer methods for the user to optionally call before mjStep is literally all that's required in order for perturbations to work, right?

I do think that it should be an explicit call though. The other modifications that the viewer makes are in mjOption, while this is mjData (xfrc_applied). Changes to mjData should not be made silently by something "passive"...

saran-t, 2023-04-14:
I've been playing some more with the API for the passive viewer and I'm really leaning towards actually applying the perturbation. Making a viewer that's truly passive is going to entail us exposing more and more events to our API surface, and user code will end up duplicating the event handling logic.

I'm now of the opinion that the "passive viewer" should in fact apply GUI inputs to the physics state. If this is not wanted then just... don't interact with the viewer!

In the next iteration, we can provide an extra option to disable interaction too.

saran-t, 2023-04-20:
The viewer enhancement that was just released as part of version 2.3.4 now applies mouse perturbation automatically as part of its `sync` operation. Please give it a try!

rtkg, 2023-04-21:
Works fine for me, thank you!