google-deepmind/mujoco #819: Mujoco-based Gym Enviroments with NoneType Exception Error in Closing Window

aalmuzairee, 2023-04-13:
Hi, 

I'm am an academic student working on research.

I have an issue similar to https://github.com/deepmind/mujoco/issues/817

I'm getting this error when rendering a mujoco-based gym environment https://github.com/Farama-Foundation/Gymnasium-Robotics in "human" mode:

Exception ignored in: <function WindowViewer.__del__ at 0x7fb4ea7d2680>
Traceback (most recent call last):
  File "miniconda/envs/gym/lib/python3.7/site-packages/gymnasium/envs/mujoco/mujoco_rendering.py", line 335, in __del__
  File "miniconda/envs/gym/lib/python3.7/site-packages/gymnasium/envs/mujoco/mujoco_rendering.py", line 328, in free
TypeError: 'NoneType' object is not callable

I'm able to render the entire scene but when the program closes I get this error. This is not a problem for a single instance run, but when I try to run something like wandb that tries to record videos every certain number of steps. All the videos are black except the very first one. My guess is there is one process being used to record videos every interval, and this exception halts the process after the first video is fully rendered. Would appreciate more insight into whether it is a glfw based issue, mujoco environment based issue, or another package, thanks!

rodrigodelazcano, 2023-04-13:
Hey @aalmuzairee can you move this issue to the Gymnasium-Robotics repository? This seems like a Gymnasium issue more than mujoco related. Thank you.

aalmuzairee, 2023-04-13:
Hi @rodrigodelazcano ,

Sure thing

closed 2023-04-13 by aalmuzairee