google-deepmind/mujoco #821: Black screen from mujoco.viewer

C-monC, 2023-04-16:
Hi,

I install mujoco using pip.
> pip show mujoco
> Name: mujoco
> Version: 2.3.3

I run the following code:
```
import mujoco
from mujoco import viewer

model = mujoco.MjModel.from_xml_path('model_testing/viscous_test.xml')
viewer.launch(model)
```
[image]

The viewer always only shows this diagonal line. I can see the actual model through the "live" pixels.

I am using ubuntu through wsl (probably not supported). Is there anyway I can see extra logs from the viewer?
I've tried adding a camera in the XML as well but it had the same result.

Here is my xml for reference:
```
<mujoco>
  <option density="1.225" viscosity="1.8e-5" wind="0 0 1" integrator="implicit"/>

  <worldbody>
    <light pos="0 0 1"/>
    <geom type="plane" size="1 1 .01" pos="0 0 -1"/>
    <body>
      <freejoint/>
      <geom type="box" size=".025 .01 0.0001" pos=".025 0 0" euler="20 0 0" mass="1e-4" fluidshape="ellipsoid"/>
      <geom type="box" size=".025 .01 0.0001" pos="-.025 0 0" euler="-19 0 0" mass="1e-4" fluidshape="ellipsoid"/>
    </body>
  </worldbody>
</mujoco>
```

yuvaltassa, 2023-04-16:
Do you get the same thing with other models? E.g. something from the `mujoco/model/` directory.

C-monC, 2023-04-16:
I just tried the adhesion example but I get the same issue. 

[image]

saran-t, 2023-04-19:
Could you please try setting the environment variables `MUJOCO_GL_DEBUG=1` and `LIBGL_DEBUG=verbose`, and see if you get any message?

C-monC, 2023-04-22:
Thanks for the help but I just switched to a Linux host and it works.

closed 2023-04-22 by C-monC

santoshchaganti, 2023-05-17:
Hi I am also facing same error. And i tried environment variables as mentioned by saran-t but no message sent back. Is there any other ways to fix this issue