google-deepmind/mujoco #822: Kinematics tree using joint `pos` tag

studywolf, 2023-04-16:
Hello,

I'm looking for some help building out the forward kinematics tree of a model I defined in XML. Here's the relevant part of the model

```
        <body name="base_link" pos="0 0 0.05">
            <geom name="link0" type="cylinder" pos="0 0 0" size="0.025 0.05" rgba="0 0 1 0.9"/>
            <inertial pos="0 0 0" mass="0" diaginertia="0 0 0" />

            <body name="link1" pos="0 0 0.05" euler="90 0 110">
                <joint name='joint0' pos="0 10 0" axis="0 0 1"/>
                <inertial pos="0 0 0.1" mass="2" diaginertia="0.226891 0.226891 0.0151074" />
                <geom name="link1" type="cylinder" pos="0 0 0.1" size="0.025 0.1" rgba="0 1 0 0.9"/>
```

The issue is the `pos` variable on the `joint` tag. If that is set to `0 0 0`, then I understand how the kinematic tree looks. When I calculate the COM of link 1 and compare it to the actual COM of link1 in Mujoco, it's very precise (1e-17 ish). 

But when the `pos` variable is set on `joint0`, things fall apart as soon as the joint angle changes. It is unclear to me how to incorporate that into the forward kinematics. Can anyone shed some light on this for me?

yuvaltassa, 2023-04-17:
It is unclear what you are trying to achieve.

What do you mean by "building out the forward kinematics tree"?

studywolf, 2023-04-17:
Sorry I was unclear! 
I would like to specify the series of symbolic transformation matrices to calculate the position of any joint, arm segment centre-of-mass, or the end-effector given all of the joint angles of the robot (like this: https://studywolf.wordpress.com/2013/08/21/robot-control-forward-transformation-matrices/).

yuvaltassa, 2023-04-17:
MuJoCo doesn't use homogenous transformation matrices internally, so I'm afraid you're on your own. Such transformations are useful for analysis but are numerically disadvantageous. Perhaps if you said *why* you are trying to compute them, we could help.

That said, the total kinematic transformation (effectively the product of the matrices you are looking for) is computed by [mj_kinematics](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mj-kinematics), it is the first function in `engine_core_smooth.c`.

If you wish to better understand this source file, which implements the smooth rigid body dynamics (excluding constraints), I recommend [Featherstone's book](https://link.springer.com/book/10.1007/978-1-4899-7560-7).

closed 2023-04-17 by yuvaltassa

studywolf, 2023-04-17:
Hello, I understand that Mujoco doesn't use the transformation matrices internally. I was hoping for was that you could provide me an intuition on how the `pos` variable on `joints` works, so that I can create them myself. As I mentioned, I've sorted out the transformation matrices for the cases where `pos` on `joints` is `"0 0 0"`, but I'm not clear on what's happening exacctly when `pos` is non-zero.

It seems like the joint and body positions are specified in the parent frame, and the body is attached to the joint such that the start location is at the specified body `pos` in the parent frame. Where this becomes especially unclear to me is when the `euler` tag is set on the body, as it seems like that orientation change takes place _before_ the joint. Are you able to offer any insight into this?

yuvaltassa, 2023-04-17:
Body position and orientation specifiers are w.r.t the parent body of this body.
Joint position and orientation specifiers are w.r.t the body in which this joint is defined.

In the case below, the joint will be be at (2, 0, 0) in the global frame.

```xml
<worldbody>
  <body pos="1 0 0">
    <joint pos="1 0 0">
```

The same applies for orientations. Generally, frame specifiers are always w.r.t to the enclosing element.

Regarding your specific question, I recommend again that you look at `mj_kinematics`. Specifically [these lines](https://github.com/deepmind/mujoco/blob/9b7d42acb8c60d9daa3c2ca87b45c0bb9f1a7318/src/engine/engine_core_smooth.c#L107) seem relevant.

Pedagogically, I think it might be valuable to create very simple models in XML, compile them to `mjModel` and understand how changes in the XML are affecting the compiled model by inspecting the `mjModel` fields. Once this makes sense to you, you can set different values of `mjData.qpos` and call `mj_kinematics`, to see how the model and the configuration interact when propagating the kinematic transformations.

studywolf, 2023-04-17:
Ah, I must have misread the documentation, I thought joints were defined wrt the parent body. I'll check out the code you referenced, thanks! 

I'm happy to report starting with a basic model and building up has been my approach; I'm not sure how much simpler it gets than a single joint XML file! :) Thanks for your reply, again. Hopefully I'll be able to sort it out from here!