google-deepmind/mujoco #823: O2MConverter (Convert OpenSim Models to MuJoCo Models)

mertalagozlu, 2023-04-17:
Hi,

I'm a student and I'm trying to use MuJoCo for academic purposes.

I'm looking for some help with O2MConverter, which enables us to convert opensim models to mujoco format. Even with many attempts, I am hving trouble converting generic wrist.osim file that can be found in opensim models using this tool ([https://github.com/aikkala/O2MConverter]. A brief summary of what I have done can be found in this stackoverflow question [https://stackoverflow.com/questions/75987855/unresolvedpackage-anaconda-conda-environment-o2mconverter](url). Briefly, I have dependency issues and cannot convert models and I thought that this can be great place to resolve the issue because it could be useful for other people. 

Thank you very much!

vikashplus, 2023-04-18:
@mertalagozlu  -- I just want to bring your attention to the **MyoSuite** project where we went thought similar pain-points. MyoSuite is an attempt at providing these solutions out of the box so that others don't have to face similar issues. Wrist model was one of the hardest to parse and stabilize, hope it helps - https://sites.google.com/view/myosuite

mertalagozlu, 2023-04-18:
Thanks for the heads up @aikkala, @vikashplus, but I am only interested in getting xml model of wrist.osim and further change it if&when necessary in the future. Is there a way to directly get the wrist model for MuJoCo without myosuite. I found models inside myosuite env as binary files, maybe I can change it as xml some way. Additionally, myosuite uses mujoco 2.1 with mujoco_py. How can I use the latest MuJoCO engine while also using myosuite. I also encountered some issues with myosuite stating such that I need to add mujoco_py to the path even though I added to the environment variables of Windows. I am using Pycharm and followed the exact instructions for installing myosuite fyi.

`conda create --name MyoSuite python=3.7.1 
conda activate MyoSuite
pip install -U myosuite`

Did not work, and not sure this is correct place to raise the question though

Pouyan97, 2023-04-19:
Myosuite is only installed on Macbook or Linux. Therefore, installing Myosuite requires Linux on Windows. Two options available are the built-in WSL or a virtual box. The virtual box performs better for Myosuite due to its simpler visualization capabilities. 

1. Install Virtual box from [Here](https://www.virtualbox.org/wiki/Downloads)
2. Follow the instructions to create a virtual box with Linux Ubuntu 20.0.04+ installed (need to download the ubuntu .iso file and include it in the creation of the VM)
3. Follow the  linux [advanced installation](https://github.com/facebookresearch/myosuite/blob/main/setup/README.md) guide to install Myosuite 
4. Download the github repo for myosuite to be able to run the python test code on your VM

**Potential issues:** 
- `IMPORT ERROR: ‘GLIBCXX_3.4.30 not found’`
Solution: solution found on stackoverflow: libstdcxx is replaced by libstdc++ in conda therefore we can need to libstdcxx-ng=12 for conda
Use the command:
`conda install -c conda-forge libstdcxx-ng=12`
If that doesn’t solve the issue refer to solutions presented [HERE](https://askubuntu.com/questions/575505/glibcxx-3-4-20-not-found-how-to-fix-this-error)

- `Illegal instructions (CORE DUMPED)` when running the test script provided in github page:
Solution: this issue might not be related to virtual box and windows itself. If you see a turtle with the V logo in the bottom right corner of your VM, these solutions and instructions can be found [HERE](https://stackoverflow.com/questions/65780506/how-to-enable-avx-avx2-in-virtualbox-6-1-16-with-ubuntu-20-04-64bit)

Pouyan97, 2023-04-19:
I have posted a solution to installing myosuite and mujoco on windows using a linux VM [HERE](https://github.com/facebookresearch/myosuite/issues/14#issuecomment-1514028917)

mertalagozlu, 2023-04-19:
Thanks @Pouyan97, I will call it as plan B if I cannot manage advanced setup in my wsl. Although I am still interested in importing .mjb models to mujoco without using myosuite. To do that, I think we need _mj_loadModel_, but mujoco.mj_loadModel does not work on loading .mjb models. In fact, it gives error as "cannot find reference in __init__.py"

So overall solution is as below to import models as .mjb format.
`model = mujoco.mj_loadModel(r'C:\ProgramData\Anaconda3\envs\MyoSuite\Lib\site-packages\myosuite\envs\myo\assets\myo_hand_pose.mjb')
`
Why do you think we can't add the model this way?

kevinzakka, 2024-01-03:
@mertalagozlu The correct API to load an MJB is:

```
import mujoco

m = mujoco.MjModel.from_binary_path("model.mjb")
d = mujoco.MjData(m)
```

Going to mark this issue as resolved for now. If you have further issues, please feel free to re-open.

closed 2024-01-03 by kevinzakka

vikashplus, 2024-01-03:
@mertalagozlu -- Please note that the binary models are MuJoCo version-specific.