google-deepmind/mujoco #826: How to use two cameras simultaneously ?

gywhitel, 2023-04-19:
Hi, I'm new to MuJoCo and I'm trying to use MuJoCo for robot simulation.

I'm looking for some help with simulation with two cameras.

I added a fixed camera like the xml below, as robot RGBD sensor.
```xml
 <body pos="0 0 1" axisangle="1 0 0 90" mocap="true">
      <geom type="box" size="0.05 0.05 0.1" rgba="0.5 0.5 0.5 0.5"/>
      <camera name="camera" mode="fixed" fovy="45"/>
    </body>
```
I managed to get color image and depth image from the camera (uisng C++), and also synthesised colorful point cloud (shown in the right snapshot) from the two images.
Yet this fixed camera causes that I cannot freely move my viewpoint in the simulation. As the left snapshot shows, I cannot move the camera view because it is fixed.
 
```cpp
 // setup camera
  mjvCamera rgbd_camera;
  rgbd_camera.type = mjCAMERA_FIXED;
  rgbd_camera.fixedcamid = mj_name2id(model, mjOBJ_CAMERA, "camera");
...
// update scene and render
    mjv_updateScene(model, data, &option, NULL, &rgbd_camera, mjCAT_ALL, &scene);
    mjr_render(viewport, &scene, &context);
```
[image]

From my understanding, the camera in `mjv_updateScene()` determines whose view is rendered later in `mjr_render()`. 
Now, I want the fixed camera as a simulation of robot visual sensor, and I also want to move the viewpoint freely so that I can check the simulation world.

Is there any way to view two cameras simultaneously, one fixed and one free? 
Or how to open two GLFW windows sharing data with each other, one for fixed sensor (the same with robot's sight), and one for free camera (explore the simulation world)?

saran-t, 2023-04-19:
Can you give a bit more context for what you're trying to do? Are you writing your own viewer from scratch? Are you trying to modify the Simulate application?

gywhitel, 2023-04-19:
Sure. I am trying to simulate a RGBD sensor in mujoco, since there seems no native RGBD sensor support in mujoco.
I select some of my codes that fetch color buffer and depth buffer from OpenGL and compute RGB point cloud from the two buffer.
```cpp
...
// setup camera
  mjvCamera rgbd_camera;
  rgbd_camera.type = mjCAMERA_FIXED;
  rgbd_camera.fixedcamid = mj_name2id(model, mjOBJ_CAMERA, "camera");
...
while (!glfwWindowShouldClose(window))
  {
    mjtNum simstart = data->time;
    while (data->time - simstart < 1.0 / 60.0)
      mj_step(model, data); // step simulation for 1/60 s
    // get framebuffer viewport
    mjrRect viewport = {0,0,0,0};
    glfwGetFramebufferSize(window, &viewport.width, &viewport.height);
    
    // update scene and render
    mjv_updateScene(model, data, &option, NULL, &rgbd_camera, mjCAT_ALL, &scene);
    mjr_render(viewport, &scene, &context);

    cv::Size img_size(viewport.width, viewport.height);
    
    // Pointers must be preallocated
    uchar* rgb_data = (uchar*) malloc(viewport.height*viewport.width * 3);
    float* depth_data = (float*) malloc(viewport.height*viewport.width * 4);
    mjr_readPixels(rgb_data, depth_data, viewport, &context);
    
    cv::Mat rgb(img_size, CV_8UC3, rgb_data);
    cv::flip(rgb, rgb, 0);

    // convert depth image from [0, 1] -> distance in meters
    cv::Mat depth(img_size, CV_32F, depth_data);
    cv::Mat depth_img_m = depth_to_meters(depth);

    //// camera intrinsics

    fovy = model->cam_fovy[rgbd_camera.fixedcamid] / 180 * M_PI / 2;
    // focal length, fx = fy
    f = viewport.height / 2 / tan(fovy);
    cx = viewport.width / 2;
    cy = viewport.height / 2;

    // visualize RGB point cloud
    *color_cloud = generate_RGB_pointcloud(rgb, depth_img_m, f, cx, cy);
...
}
```
This point cloud part works well. What bothers me is another problem.

Usually the users can move their viewpoint freely in the window, e.g. in the `simulate` program.
Yet since I used a fixed camera, I cannot do that in my program.

What I want is keeping the simulated RGBD sensor (as the code above) and having another camera to view the simulation world freely. This is basically like showing the frame of two different cameras simultaneously.

Is there any way to open two GLFW windows, one for each camera?

gywhitel, 2023-04-19:
Thanks!
I tried starting two threads, each one opening a GLFW window, which looks like  the picture below. 

Now I can view the camera image on the "Camera" window, and explore the simulation in the "simulation" window.

And by the way, we should probably also create two `mjvScene` for the two windows. Otherwise the two windows interfere with each other.
[image]

closed 2023-04-19 by gywhitel

saran-t, 2023-04-19:
OK, good to know :)

I've been spending too much time wrangling with the viewer on macOS where all GUI activities have to occur on the main thread.

gywhitel, 2023-04-21:
Sorry to hear that :(

I created a repo about simulating RGBD camera, which gives camera view, simulation world, can color pointcloud view. If you are interested, check it out [gywhitel/mujoco_RGBD](https://github.com/gywhitel/mujoco_RGBD).

[image]