google-deepmind/mujoco #827: Physical relevance of cable composite parameters

sriddle97, 2023-04-19:
I am a student working on soft robots and have been using the cable (or rope) composites to simulate flexible tubing/rods that make up our robot's structure. The model is based on a robot that has already been constructed and we would like to use it for controller validation. This would require the model to behave reasonably close to the physical robot however I'm not sure if the soft body dynamics are being modeled correctly. In particular, I am uncertain how some of the cable object parameters correlate to physical material properties.

For example, a sample xml file code (found at model/plugin/coil.xml on the mujoco GitHub page) defines the cable object as follows:

```XML
<composite prefix="actuated" type="cable" curve="cos(s) sin(s) s" count="41 1 1"
      size="0.251327412 .1 4" offset="0.25 0 .05" initial="fixed">
      <plugin plugin="mujoco.elasticity.cable">
        <!--Units are in Pa (SI)-->
        <config key="twist" value="5e8"/>
        <config key="bend" value="15e8"/>
        <config key="vmax" value="2"/>
      </plugin>
      <joint kind="main" damping="0.1" armature="0.01"/>
      <geom type="capsule" size=".005" rgba=".8 .2 .1 1"/>
    </composite>
```

The comment says the "bend" and "twist" parameters are defined in units of Pa but says nothing about what the values represent. My assumption is that "bend" correlates to the desired material's flexural stiffness, since the values are in the realm of GPa. Am I correct in assuming this? And what then do "twist" and "vmax" correlate to?

Any clarification on these would be greatly appreciated.

quagla, 2023-04-20:
Hi @sriddle97 the parameters are what you expected: 

https://github.com/deepmind/mujoco/blob/main/plugin/elasticity/cable.cc#L134

i.e. twist and bend are, respectively, the Shear and Young moduli. See here for how they are used to compute the actual stiffnesses taking into account the cross section geometry:

https://github.com/deepmind/mujoco/blob/main/plugin/elasticity/cable.cc#L197

The other parameter vmax is just a visualization parameter for setting the color scale of the stresses.

sriddle97, 2023-04-21:
@quagla Thank you for confirming this!

closed 2023-04-23 by saran-t