google-deepmind/mujoco #830: Joints attached to composite cables pulling away

sriddle97, 2023-04-21:
I am a student working on a soft robot model. I've been using cable composite objects connected at "joints" made with <equality/connects> (see images below). The hexagonal rings are circumferential muscles that contract the flexible body. For some reason when the muscles contract the sites they are connected to pull away from the joint. However, the structure is still deforming as expected, at least qualitatively. Is this a glitch in the rendering or is it a fixable issue in the model? I would like to make sure that the structure deformations I am getting are numerically accurate. When the model is read in from the xml file some of the sites visually start off the joints as well, in case that is helpful information for debugging this.

I am also curious, is it possible to define damping and stiffness in an <equality/connect> like we can in a <body/joint>?

Here are some screenshots showing what I mean:
[image]
[image]

yuvaltassa, 2023-04-21:
This looks very cool!

I don't think we can help if you don't attach the model, these images are too complicated to diagnose.

Regarding your question, yes, constraints have solimp and solref, which affect how the solver tries to resolve them. I only have a very vague sense of what you're trying to achieve, but you might want to try weld constraints with very low [torque scale](https://mujoco.readthedocs.io/en/latest/XMLreference.html#equality-weld-torquescale).

And don't forget to turn on the equality visualisation flag (works for both weld and connect) it should be helpful in understanding what's going on.

sriddle97, 2023-04-21:
Thanks!

The goal is to have the structure diameter shrink when the muscles contract to produce a lateral elongation. The white spheres are marking the sites where <connects> were used to create joints (essentially pin joints to allow rotation between cables). They should be aligned with the points at which any two cables intersect. but as you'll see some are a bit off, even at the start.

Ideally these sites should not pull away from the joint.

I have attached a zip file with the model.
[worm_3_seg_testing.zip](https://github.com/deepmind/mujoco/files/11299799/worm_3_seg_testing.zip)

sriddle97, 2023-05-03:
Hi @yuvaltassa , have you had a chance to test this model out? I'm curious to see if you'll have more luck troubleshooting this than I did.

yuvaltassa, 2023-05-07:
Finally got around to looking at this.

I'm sorry but the model is too complicated for me to understand both what is happening and what you expect to happen. Can you please make a minimal model that just shows the specific issue you are experiencing?

One thing I did notice is that all the tendons are connected to free bodies, why are they not just connected directly to the spring structure?