google-deepmind/mujoco #831: Hollow cylinder imported from stl file as a solid cylinder

sriddle97, 2023-04-21:
I am working on a robot that needs to be tested in a pipe environment.  I have been trying to import a hollow cylindrical tube from an stl file into Mujoco with little success. When the object renders it looks correct (hollow) but I am unable to pass anything through it. I used a falling ball as a test and the ball simply stops in the air where a solid cylinder's flat surface would be.

I stumbled upon a mujoco help thread that discussed approximating a hollow cylinder with solid rectangular objects oriented in a circle. I, however, cannot use this method as I will eventually need to test pipes with smooth bends and continuously varying diameters. 

Is there an easy fix to this that I am missing? Most robots have components with cylindrical cutouts or holes for bolts and axles to pass through so I would imagine there is a way to make it work but so far I am stumped.

[image]

saran-t, 2023-04-23:
At the moment, approximating the hole as a collection of boxes is the only method available. Colliding non-convex geometries is a difficult computational problem to perform in realtime (or indeed faster-than-realtime). MuJoCo currently computes convex hulls from meshes for collision purposes.

We are internally working on a few ideas for non-convex collision it's still early days and we can't promise anything yet.

sriddle97, 2023-04-23:
Alright, thank you for clarifying this. I will try looking into altering the box approximation for more complex geometries then.

closed 2023-04-24 by nimrod-gileadi

yuvaltassa, 2023-04-25:
Relatedly, there is a [new player](https://github.com/SarahWeiii/CoACD) in the Convex Decomposition field which seems quite promising. Haven't tested it myself.