google-deepmind/mujoco #832: Measuring forces in tendons

thawke, 2023-04-24:
Hello,

I want to measure the tension in a tendon attached to an otherwise free body. In lieu of a `tendonfrc` sensor, I'm trying to use force sensors at the tendon attachment sites, with dummy bodies containing the sites.

I expect to see the same force reading from both sensors, with opposite signs, but I'm getting 0 output. I've tried a few configurations and cannot manage to read tendon forces at all by this method. I have obviously misunderstood something.

[image]

```xml
<mujoco>
    <option>
        <flag gravity="enable"/>
    </option>

    <default>
        <site type="sphere" rgba="1 0 0 1" size="0.005"/>
        <tendon rgba="0 1 0 1" stiffness="500" damping="10"/>
    </default>
    
    <visual>
        <headlight diffuse=".7 .7 .7"/>
    </visual>

    <worldbody>
        <body name="slidebody" pos ="0 0 0">
            <joint name="slide" type="slide" axis="0 0 1" armature="1" damping="100" range="-0.2 0.2" limited="true"/>
            <inertial mass="0.01" pos="0 0 0" diaginertia="1 1 1"/>
            <body name="originbody" pos="0 0 0">
                <site name="origin0" pos="0 0 0"/>
            </body>
        </body>
        <body name="massbody" pos="0 0 -0.3">
            <freejoint/>
            <geom name="massgeom" type="box" size="0.05 0.05 0.05" pos="0 0 0" mass="1"/>
            <body name="insertionbody" pos="0 0 0">
                <site name="insertion0" pos="0 0 0.05"/>
            </body>
        </body>
    </worldbody>

    <tendon>
        <spatial name="rope0" springlength="0 0.25">
            <site site="origin0"/>
            <site site="insertion0"/>
        </spatial>
    </tendon>

    <sensor>
        <force name="tension0" site="origin0"/>
        <force name="tension1" site="insertion0"/>
    </sensor>

    <actuator>
        <position name="verticalpos" joint="slide" kp="10000" />
    </actuator>

</mujoco>
```

The nesting works as expected in the following example with no tendon:

```xml
<mujoco>

    <visual>
            <headlight diffuse=".7 .7 .7"/>
    </visual>

    <worldbody>
        <body name="fixed body">
            <body name="dummy body">
                <site name="force1"/>
                <body name="heavybody">
                    <inertial mass="1" diaginertia="1 1 1" pos="0 0 0"/>
                    <site name="force2"/>
                </body>
            </body>
        </body>
    </worldbody>

    <sensor>
        <force name="f1" site="force1"/>
        <force name="f2" site="force2"/>
    </sensor>

</mujoco>
```

thawke, 2023-04-25:
Thanks for your reply - be good to hear what you conclude.

> You have no misunderstanding and your expectations are valid.

^ I'm going to get this on a teeshirt

yuvaltassa, 2023-11-06:
Still not fixed, but the documentation at least [acknowledges the bug](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-rnepostconstraint). 

Hopefully we'll get to this soon.

ekchapman, 2023-11-17:
@yuvaltassa, thanks for the update. Do you know when we might expect a fix? In the meantime, I have a workaround that others may find useful. Does this workaround look reasonable?

Rather than attaching a tendon endpoint directly to a body **B**, I attach it instead to a "mounting body" **M**, which is a free body that gets welded to **B**. Now when you measure the forces acting on **B**, you measure the force due to the weld constraint, rather than (trying to) measure the force from the tendon.

You have to tune the inertial properties of the free bodies and the solver parameters of the weld, but otherwise it seems to work fine. Note that, in the example below, `sensordata[2] == 9.996` but `sensordata[5] == -9.8`. This is because the two mounting bodies each have mass `0.01`. 

```
<mujoco>
    <option gravity="0 0 -9.8">
        <flag gravity="enable"/>
    </option>

    <default>
        <site type="sphere" rgba="1 0 0 1" size="0.005"/>
        <tendon rgba="0 1 0 1" stiffness="500" damping="10"/>
    </default>
    
    <visual>
        <headlight diffuse=".7 .7 .7"/>
    </visual>

    <worldbody>
        <body name="slidebody" pos ="0 0 0">
            <joint name="slide" type="slide" axis="0 0 1" armature="1" damping="100" range="-0.2 0.2" limited="true"/>
            <inertial mass="0.1" pos="0 0 0" diaginertia="1 1 1"/>
            <body name="originbody" pos="0 0 0">
                <site name="origin0" pos="0 0 0"/>
            </body>
        </body>
        <body name="massbody" pos="0 0 -0.3">
            <freejoint/>
            <geom name="massgeom" type="box" size="0.05 0.05 0.05" pos="0 0 0" mass="1" rgba="0.5 0.5 0.5 0.3"/>
            <body name="insertionbody" pos="0 0 0">
                <site name="insertion0" pos="0 0 0.05"/>
            </body>
        </body>

        <body name="tendon_mount_0" pos="0 0 0">
            <freejoint/>
            <inertial mass="0.01" pos="0 0 0" diaginertia="1 1 1"/>
            <site name="tendon_mount_site_0" rgba="0.3 0.6 0.9 0.6" size="0.008"/>
        </body>

        <body name="tendon_mount_1" pos="0 0 -0.25">
            <freejoint/>
            <inertial mass="0.01" pos="0 0 0" diaginertia="1 1 1"/>
            <site name="tendon_mount_site_1" rgba="0.3 0.6 0.9 0.6" size="0.008"/>
        </body>

    </worldbody>

    <tendon>
        <spatial name="rope0" limited="true" range="0 0.25">
            <site site="tendon_mount_site_0"/>
            <site site="tendon_mount_site_1"/>
        </spatial>
    </tendon>

    <equality>
        <weld body1="originbody" body2="tendon_mount_0"/>
        <weld body1="insertionbody" body2="tendon_mount_1"/>
    </equality>

    <sensor>
        <force name="tension0" site="origin0"/>
        <force name="tension1" site="insertion0"/>
    </sensor>

    <actuator>
        <position name="verticalpos" joint="slide" kp="10000" />
    </actuator>

</mujoco>

```

[image]

yuvaltassa, 2024-09-04:
@thowell and I looked at this again recently. It's surprisingly tricky to implement, but at least we have a workplan 🙂

shantnavagarwal, 2024-11-26:
I tried the approach mentioned above:
> """Rather than attaching a tendon endpoint directly to a body B, I attach it instead to a "mounting body" M, which is a free body that gets welded to B."""

However, this results in some weird behaviour. I can now see my tendons stretching (acting like a spring). 
This did not happen earlier when the tendons were directly attached to the body (B).

In my environment, I am making two cylinders pull the big (red) cylinder using a string (Spatial tendon). I have set limited=True for the tendons.

Here is a video of the behaviour. Expected behaviour is that there should be no relative motion between welded bodies (For eg. the blue sphere on the red cylinder). The tendon's must not stretch either. It may be that tendon is not stretching but the weld is stretching a lot.

https://github.com/user-attachments/assets/43a10166-b509-4a2a-b629-6e99791622d3

Here is my XML file:
``` .xml
<mujoco>
  <asset>
    <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3"
     rgb2=".2 .3 .4" width="300" height="300" mark="none"/>
    <material name="grid" texture="grid" texrepeat="1 1"
     texuniform="true" reflectance=".2"/>
  </asset>

  <worldbody>
    <light name="light1" pos="5 5 20"/>
    <light name="light2" pos="-5 5 20"/>
    <light name="light3" pos="-5 -5 20"/>
    <light name="light4" pos="5 -5 20"/>

    <geom name="floor" type="plane" pos="0 0 0" size="8 8 .1" friction="0.5 0.005 0.0001" material="grid"/>
    <geom name="wall1" type="box" pos="8 0 0" size="0.1 11 2" rgba="0.5 0.5 0.5 1"/>
    <geom name="wall2" type="box" pos="-8 0 0" size="0.1 11 2" rgba="0.5 0.5 0.5 1"/>
    <geom name="wall3" type="box" pos="0 8 0" size="11 0.1 2" rgba="0.5 0.5 0.5 1"/>
    <geom name="wall4" type="box" pos="0 -8 0" size="11 0.1 2" rgba="0.5 0.5 0.5 1"/>
    <camera name="fixed" pos="0 0 25" xyaxes="1 0 0 0 1 0"/>

  <body name="goal_body">
		<joint name="goal_x_prism" type="slide" axis="1 0 0"/>
		<joint name="goal_y_prism" type="slide" axis="0 1 0"/>
    <geom name="goal_geom" type="sphere" pos="0 0 1" size="0.1" rgba="1 0.84 0 1"/>
  </body>

	<body name="square_body" pos="1 1 0">
		<joint name="square_x_prism" type="slide" axis="1 0 0"/>
		<joint name="square_y_prism" type="slide" axis="0 1 0"/>
    <joint name="square_z_prism" type="slide" axis="0 0 1"/>
		<joint name="square_z_rotat" type="hinge" axis="0 0 1"/>
		<geom name="square_geom" type="cylinder" size="0.3 0.1" pos="0 0 0.3" rgba="1 0 0 1"/>
	</body>

	<body name="circle1_body" pos="2 2 0">
		<joint name="circle1_x_prism" type="slide" axis="1 0 0"/>
		<joint name="circle1_y_prism" type="slide" axis="0 1 0"/>
		<joint name="circle1_z_prism" type="slide" axis="0 0 1"/>
		<geom name="circle1_geom" type="cylinder" size="0.1 0.3" pos="0 0 0.6" rgba="0 1 1 1"/>
	</body>

	<body name="circle2_body" pos="0 0 0">
		<joint name="circle2_x_prism" type="slide" axis="1 0 0"/>
		<joint name="circle2_y_prism" type="slide" axis="0 1 0"/>
		<joint name="circle2_z_prism" type="slide" axis="0 0 1"/>
		<geom name="circle2_geom" type="cylinder" size="0.1 0.3" pos="0 0 0.6" rgba="0 1 0 1"/>
	</body>

  <body name = "square_tendonbody" pos="1 1 0">
    <joint name="square_tendonjoint" type="free" stiffness="0" damping="0" frictionloss="0" armature="0"/>
    <geom name = "square_tendongeom" type="sphere" size="0.05" pos="0 0 0.4" rgba="0 1 1 0.5"/>
		<site name="IMU" pos="0 0 0"/>
  </body>

  <body name = "circle1_tendonbody" pos="2 2 0">
    <freejoint/>
    <inertial mass="0.01" pos="0 0 0" diaginertia="1 1 1"/>
    <site name="circle1_site" rgba="0.3 0.6 0.9 0.6" size="0.008"/>
  </body>

  <body name = "circle2_tendonbody" pos="0 0 0">
    <freejoint/>
    <inertial mass="0.01" pos="0 0 0" diaginertia="1 1 1"/>
    <site name="circle2_site" rgba="0.3 0.6 0.9 0.6" size="0.008"/>
  </body>
  </worldbody>

 <tendon>
    <spatial name="wire1" limited="true" range="0 1.5" width="0.003">
      <site site="IMU"/>
      <site site="circle1_site"/>
    </spatial>
  </tendon>
  <tendon>
    <spatial name="wire2" limited="true" range="0 1.5" width="0.003">
      <site site="IMU"/>
      <site site="circle2_site"/>
    </spatial>
  </tendon>

   <equality>
        <weld body1="square_body" body2="square_tendonbody"/>
        <weld body1="circle1_body" body2="circle1_tendonbody"/>
        <weld body1="circle2_body" body2="circle2_tendonbody"/>
  </equality>

  <actuator>
    <motor name="circle1_x_actuator" joint="circle1_x_prism" gear="500" ctrllimited="true" ctrlrange="-1.0 1.0"/>
    <motor name="circle1_y_actuator" joint="circle1_y_prism" gear="500" ctrllimited="true" ctrlrange="-1.0 1.0"/>
    <motor name="circle2_x_actuator" joint="circle2_x_prism" gear="500" ctrllimited="true" ctrlrange="-1.0 1.0"/>
    <motor name="circle2_y_actuator" joint="circle2_y_prism" gear="500" ctrllimited="true" ctrlrange="-1.0 1.0"/>
  </actuator>

  <sensor>
    <accelerometer name="accelerometer" site="IMU"/>
  </sensor>
</mujoco>
```

shantnavagarwal, 2024-11-26:
This issue of stretching is solvable to some extent by using the solver constraints.
This issue is also a good reference: https://github.com/google-deepmind/mujoco/issues/278
``` .xml
  <equality>
        <weld solref="0.0002 1" body1="square_sec_body" body2="square_tendonbody"/>
        <weld solref="0.0002 1" body1="circle1_sec_body" body2="circle1_tendonbody"/>
        <weld solref="0.0002 1" body1="circle2_sec_body" body2="circle2_tendonbody"/>
  </equality>
```

MishaTikh, 2025-08-04:
Thank you! The frc_spring worked perfectly!

closed 2026-09-17 by copybara-service[bot]

yuvaltassa, 2026-09-17:
Only 4 years late, this is now fixed