google-deepmind/mujoco #835: Texture loading error in version 2.3.4

kibeomKim, 2023-04-25:
In version 2.3.4 (latest version at this time)
I accidentally encounter below error when newly setup,
File "/home/ID/FILENAME.py", line 77, in __init__
    self.model = mujoco.MjModel.from_xml_string(xml)
ValueError: Error: could not parse OBJ file '/home/ID/ADDRESS/textured.obj'
Object name = Cookies_textured, id = 0, line = 2, column = -1
when loading YCB obj data file.

So, I have to roll back the version 2.3.3 and it did not make any error to me.

yuvaltassa, 2023-04-25:
Thank you.

Yes, this is a known bug introduced in 2.3.4, sorry. It was already fixed in 66def34e0ef39a26c7008624fe137f9f6f96e0a0, so if you build from source you should be fine.

We will be releasing 2.3.5 with this fix as soon as we can.

closed 2023-04-25 by yuvaltassa

saran-t, 2023-04-25:
We determined yesterday that there's a major bug in asset loading that only affects OBJ and PNG, an only when `mjVFS` is used. The issue has been fixed in https://github.com/deepmind/mujoco/commit/66def34e0ef39a26c7008624fe137f9f6f96e0a0.

You can try to build that commit from source, but otherwise I expect to do an emergency 2.3.5 release today.

saran-t, 2023-04-25:
This should now be fixed in the 2.3.5 release.