google-deepmind/mujoco #837: How to express the graph data structure in Mujoco

lxsy-xcy, 2023-04-26:
Hi,I'm a student and I'm trying to use MuJoCo to simulate a model.

However, I have noticed that it can only express the tree data structure with XML. For example, I can create a model with two objects using this syntax:
`
<body name="box" pos="-.1 0 .05">
        <joint type="free" name = " boxes" />
        <geom size=".02 .02 .02" class="object" rgba = "1 0 0 0.2"/>

        <body name="box2" pos=".04 0 0">
                <geom size=".02 .02 .02" class="object" rgba = "0 1 0 0.2"/>
                <site name = "box_contact" type = "box" pos="-.02 0 0" quat = "0 1 0 1" size=".02 .02 .00001" group = "5"/>
        </body>
</body>
`
It just define a model with a tree data struct. However, I wonder if we can define a model with a graph data struct which means I can put the box2 outside box1 like this and two boxes still have a connection to each other. This means we can create a connection between two node in different tree, which is graph data structure.
`
<body name="box" pos="-.1 0 .05">
        <joint type="free" name = " boxes" />
        <geom size=".02 .02 .02" class="object" rgba = "1 0 0 0.2"/>
</body>
<body name="box2" pos=".04 0 0">
        <geom size=".02 .02 .02" class="object" rgba = "0 1 0 0.2"/>
        <site name = "box_contact" type = "box" pos="-.02 0 0" quat = "0 1 0 1" size=".02 .02 .00001" group = "5"/>
</body>
`
I am sorry that I'm not a English native speaker, I hope I express my question clearly.

saran-t, 2023-04-26:
If you want the two boxes with a fixed relationship with each other then the easiest thing to do is to add two `<geom type="box">` to the same `<body>`.

More generally, you can use one of the [equality constraints](https://mujoco.readthedocs.io/en/stable/XMLreference.html#equality) to impose relationships between bodies that can't be expressed as a kinematic tree (e.g. to create [kinematic linkages](https://en.wikipedia.org/wiki/Linkage_(mechanical))).

closed 2023-04-26 by saran-t

lxsy-xcy, 2023-04-27:
Thank you very much! I have already used the easiest method you mentioned before, so if I use the weld equality constraint, can I achieve the same effect?