google-deepmind/mujoco #839: Build fails with gcc and mingw makefiles : Syntax Error

rohit-kumar-j, 2023-04-28:
Build fails with gcc and "MinGW Makefiles" as generator, at .rc file.

Log:
```bash
|| [  8%] Building RC object CMakeFiles/mujoco.dir/dist/mujoco.rc.obj
C:\TDM-GCC-64\bin\windres.exe: D:\\build\\mujoco_utility\\src\\mujoco_utility\\dist\\mujoco.rc|23| syntax error
mingw32-make.exe[6]: *** [CMakeFiles\mujoco.dir\build.make|74| CMakeFiles/mujoco.dir/dist/mujoco.rc.obj] Error 1
mingw32-make.exe[5]: *** [CMakeFiles\Makefile2|1298| CMakeFiles/mujoco.dir/all] Error 2
mingw32-make.exe[4]: *** [Makefile|145| all] Error 2
mingw32-make.exe[3]: *** [CMakeFiles\mujoco_utility.dir\build.make|84| mujoco_utility/src/mujoco_utility-stamp/mujoco_utility-build] Error 2
mingw32-make.exe[2]: *** [CMakeFiles\Makefile2|326| CMakeFiles/mujoco_utility.dir/all] Error 2
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2|333| CMakeFiles/mujoco_utility.dir/rule] Error 2
mingw32-make.exe: *** [Makefile|162| mujoco_utility] Error 2
|| Exited with code 2
```

- Include the following context:
  - Operating system: Windows 10
  - GCC : version [10.3.0](https://jmeubank.github.io/tdm-gcc/download/) 
  - MuJoCo version 2.3.5

rohit-kumar-j, 2023-04-28:
According to the trace, the "mujoco.rc" file fails at:

> VALUE "Translation", 0x0409 0x04B0

I tried converting it to:
> VALUE "Translation", 0x0409, 0x04B0

then there were multiple errors:
<details>
<summary>Trace</summary>

```bash
In function 'mj_stackAlloc':
build\mujoco_utility\src\mujoco_utility\src\engine\engine_io.c|1182 col 39| error: unknown conversion type character 'z' in format [-Werror=format=]
||  1182 |     mju_error("stack overflow: max = %zu, available = %zu, requested = %zu "
||       |                                       ^
build\mujoco_utility\src\mujoco_utility\src\engine\engine_io.c|1182 col 56| error: unknown conversion type character 'z' in format [-Werror=format=]
||  1182 |     mju_error("stack overflow: max = %zu, available = %zu, requested = %zu "
||       |                                                        ^
build\mujoco_utility\src\mujoco_utility\src\engine\engine_io.c|1182 col 73| error: unknown conversion type character 'z' in format [-Werror=format=]
||  1182 |     mju_error("stack overflow: max = %zu, available = %zu, requested = %zu "
||       |                                                                         ^
build\mujoco_utility\src\mujoco_utility\src\engine\engine_io.c|1182 col 15| error: format '%d' expects argument of type 'int', but argument 2 has type 'long long unsigned int' [-Werror=format=]
||  1182 |     mju_error("stack overflow: max = %zu, available = %zu, requested = %zu "
||       |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
||  1183 |               "(ne = %d, nf = %d, nefc = %d, ncon = %d)",
||  1184 |               d->nstack * sizeof(mjtNum), stack_available_bytes, stack_required_bytes,
||       |               ~~~~~~~~~~~~~~~~~~~~~~~~~~
||       |                         |
||       |                         long long unsigned int
build\mujoco_utility\src\mujoco_utility\src\engine\engine_io.c|1183 col 23| note: format string is defined here
||  1183 |               "(ne = %d, nf = %d, nefc = %d, ncon = %d)",
||       |                      ~^
||       |                       |
||       |                       int
||       |                      %I64d
build\mujoco_utility\src\mujoco_utility\src\engine\engine_io.c|1182 col 15| error: format '%d' expects argument of type 'int', but argument 3 has type 'size_t' {aka 'long long unsigned int'} [-Werror=format=]
||  1182 |     mju_error("stack overflow: max = %zu, available = %zu, requested = %zu "
||       |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
||  1183 |               "(ne = %d, nf = %d, nefc = %d, ncon = %d)",
||  1184 |               d->nstack * sizeof(mjtNum), stack_available_bytes, stack_required_bytes,
||       |                                           ~~~~~~~~~~~~~~~~~~~~~
||       |                                           |
||       |                                           size_t {aka long long unsigned int}
build\mujoco_utility\src\mujoco_utility\src\engine\engine_io.c|1183 col 32| note: format string is defined here
||  1183 |               "(ne = %d, nf = %d, nefc = %d, ncon = %d)",
||       |                               ~^
||       |                                |
||       |                                int
||       |                               %I64d
build\mujoco_utility\src\mujoco_utility\src\engine\engine_io.c|1182 col 15| error: format '%d' expects argument of type 'int', but argument 4 has type 'size_t' {aka 'long long unsigned int'} [-Werror=format=]
||  1182 |     mju_error("stack overflow: max = %zu, available = %zu, requested = %zu "
||       |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
||  1183 |               "(ne = %d, nf = %d, nefc = %d, ncon = %d)",
||  1184 |               d->nstack * sizeof(mjtNum), stack_available_bytes, stack_required_bytes,
||       |                                                                  ~~~~~~~~~~~~~~~~~~~~
||       |                                                                  |
||       |                                                                  size_t {aka long long unsigned int}
build\mujoco_utility\src\mujoco_utility\src\engine\engine_io.c|1183 col 43| note: format string is defined here
||  1183 |               "(ne = %d, nf = %d, nefc = %d, ncon = %d)",
||       |                                          ~^
||       |                                           |
||       |                                           int
||       |                                          %I64d
build\mujoco_utility\src\mujoco_utility\src\engine\engine_io.c|1182 col 15| error: too many arguments for format [-Werror=format-extra-args]
||  1182 |     mju_error("stack overflow: max = %zu, available = %zu, requested = %zu "
||       |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
|| cc1.exe: all warnings being treated as errors
mingw32-make.exe[6]: *** [CMakeFiles\mujoco.dir\build.make|261| CMakeFiles/mujoco.dir/src/engine/engine_io.c.obj] Error 1
mingw32-make.exe[5]: *** [CMakeFiles\Makefile2|1298| CMakeFiles/mujoco.dir/all] Error 2
mingw32-make.exe[4]: *** [Makefile|145| all] Error 2
mingw32-make.exe[3]: *** [CMakeFiles\mujoco_utility.dir\build.make|84| mujoco_utility/src/mujoco_utility-stamp/mujoco_utility-build] Error 2
mingw32-make.exe[2]: *** [CMakeFiles\Makefile2|326| CMakeFiles/mujoco_utility.dir/all] Error 2
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2|333| CMakeFiles/mujoco_utility.dir/rule] Error 2
mingw32-make.exe: *** [Makefile|162| mujoco_utility] Error 2
|| Exited with code 2
```

</details>

saran-t, 2023-04-28:
We don't have the capacity to support mingw unfortunately, but if you figure it out please feel free to send in a PR.

closed 2023-04-30 by kevinzakka