google-deepmind/mujoco #842: Building MuJoCo as static library, `mujoco.a`

aSemy, 2023-04-29:
Hi,

I'm a hobbyist and I'm trying to create [Kotlin/Native](https://kotlinlang.org/docs/native-overview.html) bindings for MuJoCo.

I'm looking for some help with building MuJoCo as a static library. Kotlin/Native can interop with C libraries, but it requires the libraries are static, not dynamic.

I have followed [the MuJoCo build instructions](https://mujoco.readthedocs.io/en/latest/programming/#building-mujoco-from-source), and I can correctly produce `libmujoco.dylib`, as well as its dependencies as static `.a` libraries.

[image]

I have tried using `mujoco.dynlib` in my Kotlin/Native code, but (as I suspected) I get an error:

```
Undefined symbols for architecture x86_64:
  "_mj_makeData", referenced from:
      _com_deepmind_mujoco_cinterop_internal_mj_makeData_wrapper26 in result.o
  "_mj_loadModel", referenced from:
      _com_deepmind_mujoco_cinterop_internal_mj_loadModel_wrapper23 in result.o
ld: symbol(s) not found for architecture x86_64
```

I have tried adjusting the CMake configuration options to enable as many of the 'static' CMake options as I could find, but it did not seem to have an effect.

```
cmake \
  /path/to/mujoco/src                  \ 
  -DBUILD_SHARED_LIBS=OFF              \
  -DLINK_APPS_SHARED=OFF               \
  -DLINK_APPS_SHARED:BOOL=OFF          \
  -DBUILD_SHARED_LIBS:BOOL=OFF         \
  -DBUILD_STATIC_LIBS:BOOL=ON          \
  -DCMAKE_BUILD_TYPE:STRING=Release    \
  -DMUJOCO_BUILD_EXAMPLES:BOOL=OFF     \
  -DMUJOCO_BUILD_TESTS:BOOL=OFF        \
  -DBUILD_TESTING:BOOL=OFF             \
  -DMUJOCO_TEST_PYTHON_UTIL:BOOL=OFF   \
  -DMUJOCO_BUILD_SIMULATE:BOOL=OFF     \
  -DBUILD_STATIC_LIBS:BOOL=ON          \
  -DCMAKE_EXE_LINKER_FLAGS:STRING=-Wl  \
  -DMUJOCO_HARDEN:BOOL=ON              \
  -Wno-dev
```

I have also tried the suggestion in [this SO answer](https://stackoverflow.com/questions/43772887/cmake-creating-a-static-library) editing the `CMakeLists.txt` so that instead of `SHARED`

https://github.com/deepmind/mujoco/blob/15e77b13dd87e1f28210d109a8de7dff3583999d/CMakeLists.txt#L75

it uses `STATIC`

```
add_library(mujoco STATIC ${MUJOCO_RESOURCE_FILES})
```

but then I get an error

```
CMake Error: install(EXPORT "mujoco" ...) includes target "mujoco" which requires target "lodepng" that is not in any export set.
```

Is there a way I can compile MuJoCo as a static library, `mujoco.a`?

As mentioned, I am a C noob so perhaps this question is not appropriate here. However, I have created Kotlin/Native bindings for other libraries (RocksDB, Raylib, fswatch), and they provide options for compiling a static library. I hope that a similar option for MuJoCo is possible.

saran-t, 2023-04-29:
It's fairly unusual to build cross-language bindings as a static library. Can you explain why you need it to be static?

aSemy, 2023-04-29:
I assumed that Kotlin/Native required a static library, but I realise now I was wrong! Thanks for the hint @saran-t.

For reference, here's the `mujoco.def` file ([which is used by Kotlin/Native](https://kotlinlang.org/docs/native-c-interop.html)) I created that seems to work. It's a bit hacky, and I think there are some redundant options in there, but at least I've got a basic K/N program that seems to work.

```properties
package = com.deepmind.mujoco.cinterop.internal

headers = mujoco/mujoco.h

staticLibraries = libccd.a                \
                  liblodepng.a            \
                  libqhullstatic_r.a      \
                  libtinyobjloader.a      \
                  libelasticity.dylib     \
                  libmujoco.dylib         \
                  libmujoco.2.3.5.dylib   \
                  libtinyxml2.a

libraryPaths = /path/to/mujoco/lib/

compilerOpts = -Iinclude \
              -I/path/to/mujoco/lib/

linkerOpts = -L/path/to/mujoco/lib/ \
             -rpath /path/to/mujoco/lib/ \
             -lmujoco.2.3.5
```

The directory `/path/to/mujoco/lib/` contains

* `libccd.a`
* `libelasticity.dylib`
* `liblodepng.a`
* `libmujoco.2.3.5.dylib`
* `libmujoco.dylib`
* `libqhullstatic_r.a`
* `libtinyobjloader.a`
* `libtinyxml2.a`

closed 2023-04-29 by aSemy