google-deepmind/mujoco #843: Cylinder wrapping bug

vikashplus, 2023-04-30:
I find myself in a case where I believe that the wrapping with cylindrical object isn't performing as expected. I'm attaching a simplified model here that can help reproduce the behavior.

Steps to reproduce - 
1. Load the model into simulate visualizer (ensure tendon rendering is enabled)
2. Pause the simulate 
3. reset to key 0 
4. Move the joint slides just small mounts to see the tendon wrapping jump

```xml
<mujoco model="Cylinder wrapping bug">
    <visual>
        <headlight ambient="0.5 0.5 0.5" diffuse="0.5 0.5 0.5" />
    </visual>
    <worldbody>

        <body name="pelvis"   pos="0 0 0" quat="0.707107 0.707107 0 0">
            <inertial pos="-0.07 -0.03 0" mass="10.96" diaginertia="0.0622075 0.0532711 0.0299242" />
            <geom name="sacrum" type="cylinder" fromto="0 0 0 -0.056276 -0.07849 -0.07726" size="0.03"/>

            <!-- BUG: Cylinder as wrapping object -->
            <geom name="IL_at_brim_l_wrap" rgba="0.5 0.5 0.9 0.5" size="0.0549 0.05" pos="-0.071 -0.065 -0.0756" quat="0.874257 0.193832 0.0376181 0.443495" type="cylinder"/>

            <!-- Mostly works with sphere as a wrapping object -->
            <!-- <geom name="IL_at_brim_l_wrap" rgba="0.5 0.5 0.9 0.5" size="0.0549 0.05" pos="-0.071 -0.065 -0.0756" quat="0.874257 0.193832 0.0376181 0.443495" type="sphere"/> -->

            <site name="iliacus_l-P2" pos="-0.0135 -0.0557 -0.0756" />
            <site name="IL_at_brim_l_site_iliacus_l_side" pos="-0.00526034 -0.0756276 -0.0828915" rgba="1 1 0 1" size="0.01"/>

            <body name="femur_l" pos="-0.056276 -0.07849 -0.07726">
                <inertial pos="0 -0.195 0.0005" quat="0.7062 -0.708013 0 0" mass="8.4" diaginertia="0.1694 0.1694 0.0245269" />
                <joint name="hip_flexion_l" pos="0 0 0" axis="0 0 1" range="-0.523599 2.0944" />
                <joint name="hip_adduction_l" pos="0 0 0" axis="-1 0 0" range="-0.872665 0.523599" />
                <joint name="hip_rotation_l" pos="0 0 0" axis="0 -1 0" range="-0.698132 0.698132" />
                <geom name="l_femur" type="box" size=".01 .2 0.01" pos="0 -.25 0"/>
                <site name="iliacus_l-P3" pos="-0.0023 -0.0565 -0.0139" />
            </body>
        </body>
    </worldbody>

    <tendon>
        <spatial name="iliacus_l_tendon" springlength="0.0961207" rgba="1 0 0 1">
            <site site="iliacus_l-P2" />
            <geom geom="IL_at_brim_l_wrap" sidesite="IL_at_brim_l_site_iliacus_l_side" />
            <site site="iliacus_l-P3" />
        </spatial>
    </tendon>

    <!-- Reset to this keypose and then move the slides to see the tendon jump -->
    <keyframe>
        <key qpos='-0.366544 -0.774959 0.13962'/>
    </keyframe>

</mujoco>
```

cc: @HuaweiWang

yuvaltassa, 2023-05-01:
1. There is no bug. One of your tendon end sites is moving *inside* the wrapping cylinder. Move this site a bit down the box geom and the issue goes away.
2. Please send us loadable models. I had to add `<compiler autolimits="true" angle="radian"/>` for this model to load correctly.

closed 2023-05-01 by yuvaltassa

vikashplus, 2023-05-01:
Thanks @yuvaltassa. Confirming that it resolves the issue.