google-deepmind/mujoco #844: Get Model and Data Sizes

rohit-kumar-j, 2023-05-01:
Hi,
I'm a student and I'm trying to use MuJoCo for robotics simulation. When looking for ways to get the total size allocated by mujoco when we load an xml file, I came across the `mj_sizeModel()` function that returns the size of model. Is there a similar way to return the total size of mjData? 
My intention is to know how much cache memory we have for allocation of data intended for storing the states of the robot at each timestep...

PS: ...and if anyone has suggestions for laying out buffers per thread in memory, followed by concatenation of chunks of per thread memmory back to main memory.

saran-t, 2023-05-04:
Are you looking for the _total_ memory footprint occupied by `mjData`? If so then it's `sizeof(*d)` for the mjData itself, plus `d->nbuffer` for the amount of memory allocated for `d->buffer`, plus `d->nstack * sizeof(mjtNum)` for the amount of memory allocated for `d->arena`.

rohit-kumar-j, 2023-05-05:
Thank you for the reply.
Yes, I was looking for the Total size of mjData(+buffer+arena) .

I was wondering, why do both the small and the large models have the same mjData size?

yuvaltassa, 2023-05-05:
I presume you mean to ask why `sizeof(mjData)+buffer+arena` is the same, since `sizeof(mjData)` is constant for obvious reasons. 

The answer is that MuJoCo allocates `mjData` very generously, by default (about 13MB, IIRC). You can see how much of this memory is actually used by loading your model in simulate and looking at the info panel (F2). In order to decrease or increase the allocation, use `<size memory>`, as documented [here](https://mujoco.readthedocs.io/en/latest/XMLreference.html#size-memory) and [here](https://mujoco.readthedocs.io/en/latest/modeling.html#csize).

closed 2023-05-05 by yuvaltassa

rohit-kumar-j, 2023-05-05:
Thanks this is just what I was looking for!