google-deepmind/mujoco #845: IOException: Error loading the model: XML Error: problem reading attribute 'range' Element 'joint', line 1

kaizu3, 2023-05-01:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a (student / professor / engineer) and I'm trying to use MuJoCo for a project.

I'm looking for some help with this error when I import the humanoid mujoco scene or any mujococ scene in unity I get this error ""IOException: Error loading the model: XML Error: problem reading attribute 'range' Element 'joint', line 1 "".

Here is a screenshot / video, illustrating my question:
[image]

Please

erez-tom, 2023-05-01:
If you input a path to the text box called "Debug File Name" (at the top of the "Mj Global Settings" component) you can save the XML that's generated from your scene.  You can then inspect it (in particular, search for mentions of "range") and see which element is causing the problem.  Is this a fresh import from the humanoid XML, or a scene/prefab you've been modifying?

kaizu3, 2023-05-02:
It is a fresh import and i tried multiple models from different sources and versions of mujoco and i get the same error.
I tried it in mujoco and it worked just fine.
I linked the xml debug file if you can see any problem.
Thank you very much.

[HUMANOID.txt](https://github.com/deepmind/mujoco/files/11368750/HUMANOID.txt)