google-deepmind/mujoco #847: Why my model's physics.data.ncon is always zero?

Uptomylimit, 2023-05-02:
Hi,

I'm a student and I'm trying to use MuJoCo for simulation.

I'm looking for some help with the collision computation of mesh.

The xml is as follows:

<mujoco model="...">
    <compiler angle="radian" meshdir="..." />
    <size njmax="500" nconmax="100" />
    <option integrator="RK4" timestep="0.002" collision="all"/>
    <visual>
      <global offheight="2560" offwidth="2560"/>
    </visual>
    <default>
        <joint type="hinge" limited="true" range="-1.57 1.57" frictionloss="0.1" damping='0.5'/>
        <motor  ctrllimited="true" ctrlrange="-1. 1." forcelimited="true" forcerange="-1 1"/>
        <default class="snake">
            <geom type="mesh" conaffinity="0" contype="1" friction="1 0.005 0.0001" margin="0.001" ></geom>
            <default class="head">
                <geom rgba="1 1 1 1" mesh="head"></geom>
            </default>
            <default class="geom_odd">
                <geom rgba="1 1 1 1" mesh="body_odd"></geom>
            </default>
            <default class="geom_even">
                <geom rgba="1 1 1 1" mesh="body_even"></geom>
            </default>
            <default class="tail">
                <geom rgba="1 1 1 1" mesh="tail"></geom>

    </default>
    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/>
        <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>  
        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
        <mesh name="head" file="Base_link.STL" />
        <mesh name="body_odd" file="body1.STL" />
        <mesh name="body_even" file="body2.STL" />
        <mesh name="tail" file="body16.STL" />
    </asset>
    <worldbody>
        <geom name="floor" pos="0 0 0" size="100 100 1" type="plane" material="matplane" margin="0.001" condim="3" friction="10 10 10"/>
        <geom name="branch" pos="-1.365 -0.084 1.5" type="cylinder" size="0.05 1 0" euler="0 1.570796 0" rgba="1 0 0 1" friction="1 0.005 0.0001"/>
        <camera name="camera" mode="trackcom" pos="-1.365 -2 1.2" euler="1.570796 0 0"/>
        <body name="Base_link" pos="-0.86641004 -0.01086565  1.3357951" quat="-0.71272223  0.09020974 -0.69314648  0.05862742">

            <geom class="head" />
            <body name="body1" pos="2.22044605e-16  8.32667268e-17 -2.00000000e-03" quat="7.58165369e-01  0.00000000e+00 -1.59594560e-16  6.52062323e-01">
                <camera name="track" mode="trackcom" pos="0 -5 5"  xyaxes="1 0 0 0 1 1"/>
                <inertial pos="-0.0215451 -1.13714e-08 -0.0259557" quat="0.498731 0.501266 0.501264 0.498732" mass="0.238862" diaginertia="0.000165831 0.000145855 0.000102879" />

                <geom class="geom_odd" />
                <body name="body2" pos="-0.091 -0.026 -0.026" quat="0.70569435  0.70569734 -0.04464733  0.04464714">
                    <inertial pos="-0.0215451 -1.13682e-08 -0.0259557" quat="0.498731 0.501266 0.501264 0.498732" mass="0.238862" diaginertia="0.000165831 0.000145855 0.000102879" />

                    <geom class="geom_even" />
		...
		...
                </body>
            </body>
        </body>
    </worldbody>

<actuator>
        <motor name="L17"  joint="L17" />
        <motor name="L18"  joint="L18" />
        <motor name="L19"  joint="L19" />
        <motor name="L20"  joint="L20" />
        <motor name="L21"  joint="L21" />
        <motor name="L22"  joint="L22" />
        <motor name="L23"  joint="L23" />
        <motor name="L24"  joint="L24" />
    </actuator>
</mujoco>

I run the simulate.exe in mujoco\bin, and get the picture as follows:

[image]
2-4193-9b98-e28406ae10f4.png)

Why there isn't any contact point ? And i print physics.data.nconv, which is awlays zero, why? Can you help me ?

yuvaltassa, 2023-05-02:
Please attach a **loadable** model, XML or zip file if your model requires binary assets.

Uptomylimit, 2023-05-03:
[xml.zip](https://github.com/deepmind/mujoco/files/11381933/xml.zip)
This is a loadable model.

yuvaltassa, 2023-05-04:
The coiled part of your robot has no dofs, its geoms are considered part of the world. Geoms in the same body cannot move relative to each other and therefore cannot collide. If I grab the jointed part of your model and drape it over the pole, I get all the contacts I expect.

closed 2023-05-04 by yuvaltassa

Uptomylimit, 2023-05-04:
I guess that what you said is about the collision of the pole and the body, but I want to know why there is no collision between bodies like this picture. 
[image]
The red part is which makes me confused. They are geoms in different body, and their bodies have dofs.

yuvaltassa, 2023-05-04:
You are telling them not to collide in this line
```xml
<geom type="mesh" conaffinity="0" contype="1" friction="1 0.005 0.0001" margin="0.001" >
```
Read [this](https://mujoco.readthedocs.io/en/latest/computation.html?highlight=contype#selection) for details.

Uptomylimit, 2023-05-04:
Thank you very much!  Your answer solves my problem absolutely!
Salute!