google-deepmind/mujoco #848: Unity Plugin: Failed to Import MJCF file

ZhiquanW, 2023-05-03:
Hi,
I'm looking for some help with mujoco unity plugin.
I am working on MacOS with Univty 2021.3.18f1 (Apple Silicon) with mujoco 2.3.5.
I followed the instruction on Unity-Plugin: https://mujoco.readthedocs.io/en/2.3.5/unity.html. Download and run mujoco app, import the unity plugin in, and rename the dylib.
[image]

- The mjcf file I try to import is Ant.xml: https://github.com/iandanforth/mjcf/blob/master/ant.xml
- The error message I got is 
[image]

Could anyone give me any hints about how to fix it? 
Thank you so much.

Balint-H, 2023-06-12:
Was the library imported to the Unity project? (Can you find it in the MuJoCo section of the packages in the project explorer?)

kevinzakka, 2024-08-30:
Closing due to inactivity.

closed 2024-08-30 by kevinzakka