google-deepmind/mujoco #849: Illegal Instruction (core dumped)

lzhhh93, 2023-05-04:
Following along with https://github.com/openai/mujoco-py on Windows 10 running an Ubuntu 20.04 (64-bit) VM I run into the error:

[image]

Here is the output of cat /proc/cpuinfo: (given that 6 processor are set in my VM, only the info of the first one is posted here)
[image]

Since I have actually got mujoco-py to work using another Ubuntu 20.04 VM in my curent laptop (that one is unfortunately toasted), I believe my hardware should be fine to set up mujoco/mujoco-py.

On the other hand, if I want to get mujoco-py to work, how should I apply the solution from https://github.com/deepmind/mujoco/issues/487 ? In particular, how to build mujoco-py from source with AVX disabled? Or should I build mujoco from source with AVX disabled, then somehow get it linked to mujoco-py? A specific instruction will be greatly appreciated.

Appreciate for some help.

Sincerely.

saran-t, 2023-05-04:
You should build MuJoCo without AVX following Nimrod's instructions in #487.

We don't support mujoco-py since it's not a something that we develop our maintain. However, I expect that you should be able to follow their installation instructions normally with your custom-built copy of MuJoCo.

lzhhh93, 2023-05-04:
> You should build MuJoCo without AVX following Nimrod's instructions in #487.
> 
> We don't support mujoco-py since it's not a something that we develop our maintain. However, I expect that you should be able to follow their installation instructions normally with your custom-built copy of MuJoCo.

Thanks for a quick response. Since mujoco-py only supports Mujoco upto 2.1.0 version, could you also share the way to deactivate AVX there for Mujoco 2.1.0?

yuvaltassa, 2023-05-04:
Did you try the same instructions for 2.1.0?

saran-t, 2023-05-04:
@yuvaltassa There's no released source code for 2.1.0.

lzhhh93, 2023-05-04:
> @yuvaltassa There's no released source code for 2.1.0.

Is there any possible way that I could get over this error? Since the project depends on mujoco-py, I can only stay at the Mujoco 2.1.0.

yuvaltassa, 2023-05-04:
I'm afraid not, other than [switching away from mujoco-py](https://mujoco.readthedocs.io/en/latest/python.html#migration-from-mujoco-py) or bringing this up with mujoco-py maintainers.

closed 2023-05-04 by yuvaltassa