google-deepmind/mujoco #850: Question about sensors: 'cutoff'

rohit-kumar-j, 2023-05-05:
(clarification really... :D)

From the [docs](https://mujoco.readthedocs.io/en/latest/modeling.html#csensor):
> cutoff: real, “0”
        When this value is positive, it limits the absolute value of the sensor output. It is also  used 
         to normalize the sensor output in the sensor data plots in simulate.cc.

If the value of cutoff is -ve, will the sensor output raw sensors values starting from 0 for each sensor without normalization? (i.e calibrated to 0, at whatever position it spawns from first?)

yuvaltassa, 2023-05-05:
I believe [this line](https://github.com/deepmind/mujoco/blob/b798f89212be7c9e4739832e0b47212343ab1dd8/simulate/simulate.cc#L445) and [this line](https://github.com/deepmind/mujoco/blob/b798f89212be7c9e4739832e0b47212343ab1dd8/simulate/simulate.cc#L465) answer your question?

closed 2023-05-05 by yuvaltassa

rohit-kumar-j, 2023-05-06:
Ah! yes, it acts like a clipping operator when divided by `cutoff`.
Thanks! :D