google-deepmind/mujoco #852: Error loading the model

madddz, 2023-05-06:
Hi,
I have encountered a critical error while working on a project in Unity, I am currently facing the following error message 
_**" IOException: Error loading the model: XML Error: problem reading attribute 'range' Element 'joint', line 1 “**._
I've tried the suggested help here #845  but couldn't find the cause of the problem with the "range" parameter.
This error seems to be related to loading the model (fresh import from [here](https://github.com/deepmind/mujoco/releases/tag/2.3.5) ) but I have been unable to locate the exact cause, I tried it on different xml models.
I followed the documentations for the installation.
Am working with mujoco 2.3.5, Unity 2021.3.22f1.
Thank you very much for considering my request. Looking forward to resolve this issue with your guidance.

kevinzakka, 2023-07-19:
Can you post the model you are trying to load here? Does it load in the simulate binary?

kevinzakka, 2024-01-03:
Closing due to inactivity.

closed 2024-01-03 by kevinzakka