google-deepmind/mujoco #854: How to implement a position or velocity PID controller?

jwcho0609, 2023-05-08:
I'm trying to implement a position and velocity PID controller in m=MuJoCo. Due to the SISO nature of actuators in the MuJoCo environment, I know that each component of the PID controller can be separated into 3 different actuators. While the P and D controllers can be built with the combination of the pre-defined actuator/position and actuator/velocity, how can one implement the integral controller?

And to my knowledge, the PID controller can also be implemented by a general actuator as well by using user defined callbacks mjcb_act_gain and mjcb_act_bias. How can one go about implementing these callbacks within the python binding for MuJoCo? And how can one define what the control input is for a general controller (ex. whether the control is position for position PID or velocity for PI control)

yuvaltassa, 2023-05-08:
- You can add an [integrator](https://mujoco.readthedocs.io/en/latest/computation.html#activation-dynamics) to actuators which will integrate `ctrl`, but the integral of the joint position is something you'll have to compute in user code.
- [Callbacks from python](https://mujoco.readthedocs.io/en/latest/python.html#callbacks)
- Read the relevant [docs](https://mujoco.readthedocs.io/en/latest/computation.html#actuation-model) and or the [code](https://github.com/deepmind/mujoco/blob/49066e17943fb1494ef91f3c270b945f06274de8/src/engine/engine_forward.c#L157)

closed 2023-05-08 by yuvaltassa

jwcho0609, 2023-05-09:
Thank you for the response. Regarding the first bullet point, can't I simply use the joint position as the `ctrl` to the actuator then? 

Also, per documentation, to simulate a MIMO controller, we use a combination of actuators such as a position and velocity for a PD controller. It says that the velocity actuator would take the reference velocity of the robot. Does this mean the desired velocity or the current velocity of the joint? 

If my robot were to be position controlled, wouldn't the input to the D controller be the derivative of the positional error, not the velocity, which would make it velocity controlled instead?

yuvaltassa, 2023-05-09:
> Regarding the first bullet point, can't I simply use the joint position as the ctrl to the actuator then?

Not exactly, what you want to integrate the error right. But yes, if you compute the error you can pass that and you have your I controller.

> Does this mean the desired velocity or the current velocity of the joint?

Desired.

> If my robot were to be position controlled, wouldn't the input to the D controller be the derivative of the positional error, not the velocity, which would make it velocity controlled instead?

Correct.

I strongly recommend that you make a trivial model with say one horizontal slider joint and a known mass (so everything is trivially analytical), and then experiment with this model to see that actuation behaves in the way you expect.