google-deepmind/mujoco #857: Maximize default viewing area of the python viewer.

vikashplus, 2023-05-08:
**Is your feature request related to a problem? Please describe.**
Python viewer: There is very small surface left to view the model 
- The python visualizer starts with both left and right pane open. 
- The default font size are somewhat big on my Ubuntu laptop with 4k external screen.

**Describe the solution you'd like**
- Left and right pane should start hidden/ programmatic way to start them in hidden mode 
- Any way to specify the default font size of the viewer.

yuvaltassa, 2023-05-08:
Starting the panes hidden is bad for discoverability, I don't think that's a good idea. Also hiding them is very easy.

Controlling the default font size would be valuable though. Maybe also initial window size and location? Anything else?

saran-t, 2023-05-08:
I think @vikashplus just wants an option in `launch_passive` to have the viewer start with UI hidden, which I think is a fair request.

saran-t, 2023-05-09:
PRs are also always welcome if any of your users feels strongly enough to implement these features. :)

closed 2023-10-10 by copybara-service[bot]

saran-t, 2023-10-10:
Sorry that it's taken so long to roll this out. We're planning to release a new version that incorporates a1d0cbd6543e41ebb2612fcff3b9d3b0a0c1e9ea in the middle of next week.

If you want to try this out before then, you can `pip install mujoco --pre -f https://py.mujoco.org/`.

commit a1d0cbd6 (2023-10-10): Add `show_{left,right}_ui` kwarg to `viewer.launch(_passive)`.

Fixes #857

PiperOrigin-RevId: 572333909
Change-Id: I12247fe856afddf808eb8f386a594acac47c34b5