google-deepmind/mujoco #859: Difference in Obj file behaviors

vikashplus, 2023-05-08:
Hi, 
I'm trying to use the obj files to build some models. I'm observing some difference in texturing between the behavior of the obj mesh between the mujoco(left) and other software (meshlab, right). Note the white diagonal patch. Maybe I'm missing something?

- Mujoco2.3.4
- Object meshes are exported from blender

[image][image]

[obj_difference.zip](https://github.com/deepmind/mujoco/files/11425182/obj_difference.zip)

yuvaltassa, 2023-05-09:
This is a known issue, having to do with OBJ files allowing multiple textures on a single mesh, which is currently not possible with MuJoCo. The solution is to break up the OBJ into multiple submeshes, each referencing a single texture. Happily, @kevinzakka wrote a tool called [obj2mjcf](https://github.com/kevinzakka/obj2mjcf), which does exactly this 🙂

vikashplus, 2023-05-09:
@yuvaltassa  -- the model I shared only uses a single texture (`scene.png`). No?

vikashplus, 2023-05-13:
@quagla  -- Yes editing the `vt` fixes the issue in MuJoCo.

The mesh in its original form is an export from blender. It loads back well on other softwares likes like MeshLab. What I'm trying understand is why is the behavior different in MuJoCo, while okay in others.

closed 2023-05-16 by copybara-service[bot]

yuvaltassa, 2023-05-17:
@vikashplus, thanks for the bug report!
This is now fixed at head.

commit 9db444cd (2023-05-16): Fixed texture mapping bug. Before this fix, the first texture coordinate was not correctly processed.

Fixes #859.

PiperOrigin-RevId: 532501028
Change-Id: I1fe9ca470f4d64af606479b91387e43bc56c23dc