google-deepmind/mujoco #862: C1083 Cannot open include file: 'stdalign.h': No such file or directory (Visual Studio 2019)

erwincoumans, 2023-05-11:
Compiling latest MuJoCo gives an error with Visual Studio 2019 due to a missing header file. It is only due to a single line of code. Would it be possible to replace the alignas with __declspec/__attribute__ aligned ?

```
#ifdef __cplusplus
extern "C" {
#else
//#include <stdalign.h>
#endif

#ifdef _WIN32
    #define ATTRIBUTE_ALIGNED16(a) __declspec(align(8)) a
#else
	#define ATTRIBUTE_ALIGNED16(a) a __attribute__ ((aligned (8)))
#endif

ATTRIBUTE_ALIGNED16(struct) mjCollisionTree_
{
    int node1;
    int node2;
};
```

nimrod-gileadi, 2023-05-11:
I don't have VS19 running anywhere, but this [blog post](https://devblogs.microsoft.com/cppblog/c11-and-c17-standard-support-arriving-in-msvc/) says it should support C11 and stdalign.h. Are you compiling with C11 support?

erwincoumans, 2023-05-23:
The bug wasn't fixed for my Visual Studio 2019 version, still seeing this error:
F:\develop\kit-extension\kit-extension-template-cpp\mujoco\src\engine/engine_collision_driver.h(32,1): fatal  error C1083: Cannot open include file: 'stdalign.h': No such file or directory
```
#ifdef __cplusplus
extern "C" {
#elif !defined(__STDC_VERSION__) || __STDC_VERSION__ < 201112L
// No C11 support in Visual Studio 2019 update 7 and earlier.
// However, MSVC allows C++ alignas to be used in C code, so
// we can just skip the #include <stdalign.h>.
#ifndef _MSC_VER
#error "Compiler does not support C11."
#endif
#else
#include <stdalign.h>
#endif

```

saran-t, 2023-06-12:
Are you able to check with `__STDC_VERSION__` is set to on your system? We're now running CI against VS 2019 on GitHub Actions and everything seems to be working fine:

```
-- Building for: Visual Studio 16 2019
-- The C compiler identification is MSVC 19.29.30148.0
-- The CXX compiler identification is MSVC 19.29.30148.0
```

erwincoumans, 2023-06-14:
```
#define STRING2(x) #x
#define STRING(x) STRING2(x)
#pragma message(STRING(__STDC_VERSION__))
```
and the output is
```
 201112L
```

saran-t, 2023-06-14:
OK, then MSVC is lying about it's C standard support. The compiler is claiming to be C11 compliant and so should provide `stdalign.h`.

This is in fact the case in Visual Studio 2019 on the GitHub Actions runner image too. Would you be able to compare your local version against the version number that CMake spits out in our CI log?

saran-t, 2023-06-14:
OK, then MSVC is lying about it's C standard support. The compiler is claiming to be C11 compliant and so should provide `stdalign.h`.

This is in fact the case in Visual Studio 2019 on the GitHub Actions runner image too. Would you be able to compare your local version against the version number that CMake spits out in our CI log?

Can you check the value of `_MSC_FULL_VER`  as well?

traversaro, 2023-06-21:
I am experiencing the same problem on GiHub Actions, see https://github.com/traversaro/mujoco-conda-forge-ci/actions/runs/5337234014/jobs/9673167210 .  In my case I have:

~~~
_MSC_VER=1929
_MSC_FULL_VER=192930148
__STDC_VERSION__=201112L
~~~

saran-t, 2023-06-21:
Ugh, at this point maybe we should just assume that this header never exists on _MSC_VER <= 1929. Mind trying this and sending a PR @traversaro ?

traversaro, 2023-06-22:
> Ugh, at this point maybe we should just assume that this header never exists on _MSC_VER <= 1929. Mind trying this and sending a PR @traversaro ?

Unfortunatly the issue was not in the compiler version, but in the Windows SDK version, see https://github.com/deepmind/mujoco/pull/946 for a discussion and a (possible, perhaps not ideal) solution.

traversaro, 2023-06-22:
> Unfortunatly the issue was not in the compiler version, but in the Windows SDK version, see #946 for a discussion and a (possible, perhaps not ideal) solution.

Actually, the proposed solution does not work. I would just suggest to people with this problem to install a recent Windows SDK and specify it with `CMAKE_SYSTEM_NAME` to the CMake build, or to update to a newer Windows version so  that CMAKE_SYSTEM_NAME will default to a recent enough Windows SDK version.

erwincoumans, 2023-08-15:
I'm still having the error with latest version of Visual Studio 2022 installed (Microsoft Visual Studio Community 2022 (64-bit) - Current Version 17.4.4).  Its default Windows SDK version is 10.0.19041.0.

```
 I added some CMake + C code that does not include stdalign.h if the Win SDK version is <= 10.0.22000 .
```
This would be a more user-friendly solution, as I suggested in the initial post using  __declspec align(16).

Otherwise, it would be good to point users that MuJoCo on Windows requires Visual Studio 2019 or later, and run the Visual Studio Installer, click 'Modify' and make sure the Windows SDK version 10.0.22000 or later is installed (and de-select older versions).

MSVC 2022
[image]
MSVC 2019
[image]

saran-t, 2023-08-29:
@erwincoumans @traversaro could you help me confirm if 4c899b92cc3b0ba6a8a9e78ccc9e18cb7485b743 can now be built on older MSVC+Windows SDK combinations?

traversaro, 2023-08-30:
> @erwincoumans @traversaro could you help me confirm if [4c899b9](https://github.com/deepmind/mujoco/commit/4c899b92cc3b0ba6a8a9e78ccc9e18cb7485b743) can now be built on older MSVC+Windows SDK combinations?

I just tested on the CI that was triggering the error for me, and indeed that error is still present in mujoco 2.3.6 while it is fixed by using https://github.com/robotology/robotology-superbuild/pull/1465, see https://github.com/robotology/robotology-superbuild/pull/1465 for more details.

flavienbwk, 2024-01-03:
I just typed this error and found this issue. [Upgrading the Windows 10](https://github.com/microsoft/vcpkg/issues/25654#issuecomment-1180039747) kit helped me.

commit 133b8d09 (2023-05-11): Make MuJoCo buildable with Visual Studio 2019 update 7 and earlier.

Also enable MSVC 2019 builds on GitHub Actions.

Fixes #862.

PiperOrigin-RevId: 531167754
Change-Id: I17687fb287c38a7f2245c6f7ba59a22b772e3a8c
commit d1d79f39 (2023-08-28): Mention Windows build issue regarding missing C header file.

Fixes #862

PiperOrigin-RevId: 560712857
Change-Id: I530f102b645d7d7b8dd168349162a2d38f3ad9be