google-deepmind/mujoco #864: Unity imported menagerie model shaky behaviour

Norite-dev, 2023-05-12:
Hi,

I'm an engineer and trying to use MuJoCo with Unity for education.
i've managed to import UR5e robot model from your great mujoco-menagerie collection.
OS: macOS Ventura 13.3.1
CPU: Intel Core i5
Unity: 2021.3.23f1
Mujoco: 2.3.5

It works, but behaves so shaky, can you please tell why is it so? And maybe how to fix it?
Here is the [video](https://github.com/deepmind/mujoco/assets/81800286/df516a49-2acc-4c68-9093-4dd7a0b1ed79).

Thank you guys,

kevinzakka, 2023-05-12:
Hi @Norite-dev, did you import the latest UR5e, i.e. as of [this commit](a15682c976db0cafc6c0ac8524e22cb67d14bef6)?

kevinzakka, 2023-05-12:
If you are using the Euler integrator, I would recommend you switch to the previous actuator / joint definition, specifically [this commit](https://github.com/deepmind/mujoco_menagerie/blob/ec982782ddc3b63a41cd9010e406ed272ac49a31/universal_robots_ur5e/ur5e.xml).

Norite-dev, 2023-05-12:
@kevinzakka thank you! Way less shaky with latest commit. [Still some disturbance ](https://github.com/deepmind/mujoco/assets/81800286/4282569e-84ce-4780-b825-3bf31cf1d111) on small/last joints, and btw works only with global_settings switched off. 
Thanks again!

kevinzakka, 2023-05-15:
Should be fixed by 31c265782b2931963173b1fd94acb2ec01fc0c95.

closed 2023-05-15 by kevinzakka