google-deepmind/mujoco #867: Simulating sand filled balls

mgomezandreu, 2023-05-13:
Hi,

I'm a student and I'm trying to use MuJoCo to simulate a specific type of juggling ball. 
It is made from hard plastic and partly filled with sand which gives it non standard properties.

During collisions it exhibits a non bouncing behavior much like a soft ball.

 <default class="ball">
            <geom type="sphere" size="0.0375" mass="0.067" contype="1" conaffinity="0" group="2" rgba="0.8 0.15 0.15 1" solref="-60000 -400" condim="6" friction="0 0 0"/>
        </default>

When accelerated in contact with a rigid body it behaves more like a harder ball reacting to the acceleration without deformation. This is important because deformation can mess up the subsequent motions direction and speed.

 <default class="ball">
            <geom type="sphere" size="0.0375" mass="0.067" contype="1" conaffinity="0" group="2" rgba="0.8 0.15 0.15 1" solref="-600 -400" condim="6" friction="0 0 0"/>
        </default>

Using the provided XML snippets (which only differ in the solref property) both desired properties can be archived separately, but not at the same time.
Since I need the ball to be stopped by collision without bouncing and to be accelerated without deformation, I am searching for a way to transition/switch between both properties depending on what the system is currently doing.

I would be thankful for some tips!

Best regards,
Mario

yuvaltassa, 2023-05-18:
A simple approximation is to _model the sand_ as "ballast" inside the ball, connected with e.g. a frictional ball joint. 

See model below. 

https://github.com/deepmind/mujoco/assets/6387526/64067ef3-91d6-4339-85e0-491d72cc7149

([YouTube link](https://youtu.be/znJcnwnIdmg))

<details><summary> model of ball filled with "sand" </summary>

```xml
<mujoco model="sand-ball">

  <asset>
    <texture name="ball" type="cube" builtin="flat" mark="cross" width="13" height="13" rgb1="0.2 0.2 0.3" rgb2="0.2 0.2 0.3" markrgb="1 1 1"/>
    <material name="ball" texture="ball" texuniform="true" rgba="1 1 1 0.3"/>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1=".1 .2 .3" rgb2=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="5 5" texuniform="true" reflectance=".2"/>    
  </asset>

  <visual>
    <headlight ambient="0.3 0.3 0.3"/>
    <map stiffness="3000"/>
  </visual>

  <default>
    <geom condim="6" friction="0.8 0.004 0.006"/>
  </default>

  <option density="1.225" viscosity="1.8e-5"/>

  <worldbody>
    <geom type="plane" size="2 2 .02" material="grid"/>
    <body name="sphere" pos="0 0 .2">
      <light pos="0 0 1" mode="track"/>
      <freejoint/>
      <geom size=".04" mass=".02" material="ball"/>
      <body name="sand">
        <joint type="ball" frictionloss=".015"/>
        <geom size=".02" rgba="0.7 0.7 0 1" mass=".2" pos="0 0 -.02" contype="0" conaffinity="0"/>
      </body>
    </body>
  </worldbody>

</mujoco>
```
</details>

closed 2023-05-18 by yuvaltassa

kevinzakka, 2023-05-19:
@yuvaltassa you never cease to amaze me 🤯

mgomezandreu, 2023-05-24:
Thanks a lot for the quick and detailed reply. I will try it out!

skylernov, 2023-10-12:
I am just beginning with Mujoco and am kind of lost, how did you create a hollow sphere with an object inside?

saran-t, 2023-10-12:
The outer sphere isn't hollow. The "sand" is a direct descendent of the body containing the outer sphere so [no contacts are generated](https://mujoco.readthedocs.io/en/stable/computation.html#selection) between them.

yuvaltassa, 2023-10-12:
@skylernov note the model is in my comment above