google-deepmind/mujoco #873: Model documentation of cable composite object missing

giotherobot, 2023-05-16:
Hi,

I'm a robotics engineer trying to use MuJoCo for simulating a flexible thin metal rod.

In the model documentation of the composite object there is no mention of the cable: https://mujoco.readthedocs.io/en/latest/modeling.html#composite-objects, even though trying to use a rope object triggers a warning that it is deprecated in favor of the cable.

The description in the XML reference is there and very useful: https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-composite

It would be useful to have a more complete description of the cable composite object.

quagla, 2023-05-16:
I will write the documentation, thank you for flagging this.

In the meantime, the following discussions might be useful:
https://github.com/deepmind/mujoco/issues/827
https://github.com/deepmind/mujoco/issues/727

Any questions, please let me know.

giotherobot, 2023-05-16:
Thanks for the feedback! I will look into those.

commit 139a8ae2 (2023-10-17): Add flex and elasticity documentation. Fixes #873.

PiperOrigin-RevId: 574179040
Change-Id: Iaf90bbf2b2c3fe075c940c2b66b7b841f256e653
closed 2023-10-17 by copybara-service[bot]