google-deepmind/mujoco #874: unused variables causing compile errors (warning as errors)

erwincoumans, 2023-05-16:
There are a few unused variables in source files on Ubuntu 20.04 with g++ (Ubuntu 7.5.0-6ubuntu2) 7.5.0
Trivial to fix of course:

```
~/dev/omni_mujoco/kit-extension-template-cpp/mujoco: git diff *cc
diff --git a/simulate/simulate.cc b/simulate/simulate.cc
index 945058d..c49ca62 100644
--- a/simulate/simulate.cc
+++ b/simulate/simulate.cc
@@ -1056,7 +1056,9 @@ void UpdateSettings(mj::Simulate* sim, const mjModel* m) {
 int ComputeFontScale(const mj::PlatformUIAdapter& platform_ui) {
   // compute framebuffer-to-window ratio
   auto [buf_width, buf_height] = platform_ui.GetFramebufferSize();
+  (void) buf_height;
   auto [win_width, win_height] = platform_ui.GetWindowSize();
+  (void) win_height;
   double b2w = static_cast<double>(buf_width) / win_width;
 
   // compute PPI
diff --git a/src/xml/xml_native_reader.cc b/src/xml/xml_native_reader.cc
index ed0b77f..d0585b3 100644
--- a/src/xml/xml_native_reader.cc
+++ b/src/xml/xml_native_reader.cc
@@ -2216,6 +2216,7 @@ void mjXReader::Extension(XMLElement* section) {
 
       bool already_declared = false;
       for (const auto& [existing_plugin, existing_slot] : model->active_plugins) {
+        (void) existing_slot;
         if (plugin == existing_plugin) {
           already_declared = true;
           break;
diff --git a/src/xml/xml_native_writer.cc b/src/xml/xml_native_writer.cc
index b4c2105..a2698a7 100644
--- a/src/xml/xml_native_writer.cc
+++ b/src/xml/xml_native_writer.cc
@@ -1144,6 +1144,7 @@ void mjXWriter::Extension(XMLElement* root) {
 
   // write <plugin> elements for plugins without explicit instances
   for (const auto& [plugin, slot] : model->active_plugins) {
+    (void) slot;
     if (seen_plugins.find(plugin) == seen_plugins.end()) {
       plugin_elem = InsertEnd(section, "plugin");
       WriteAttrTxt(plugin_elem, "plugin", plugin->name);

```

yuvaltassa, 2023-05-19:
Thanks @erwincoumans!
Can you send this as a PR please?

saran-t, 2023-05-22:
I remember seeing this and after some digging found that it's logged as a [GCC bug](https://gcc.gnu.org/bugzilla/show_bug.cgi?format=multiple&id=81767) which was fixed in GCC 8.

I'm of the opinion that we should minimise code noise that's only there to silence buggy compiler diagnostics. A better option would be to just exclude `-Wunused` (specified [here](https://github.com/deepmind/mujoco/blob/main/cmake/MujocoOptions.cmake#L92)) when building with GCC 7 or earlier.

closed 2024-01-08 by yuvaltassa