google-deepmind/mujoco #878: Changing worldbody coordinate system

areiner222, 2023-05-20:
Hi,

I'm a data scientist and I'm trying to use MuJoCo for human body simulation.

I'm looking for some help with being able to change the coordinate system at the worldbody level  (right hand coord with y down) to coincide with my input data. Is this possible?

Thanks!
Alex

yuvaltassa, 2023-05-20:
This is not possible. 

You can either 
- Do the swizzling in your code.
- Set gravity to point in the negative Y and suffer a less than ideal interactive vis experience.
- Add a static body to your world with the frame that you want and then children of that body would use that as their local frame. Note that this may not solve your problem since free joints are only allowed in the world body. You could get around that restriction by defining your own free joints as 3 sliders and a ball.

I would personally recommend [swizzling](https://en.wikipedia.org/wiki/Swizzling_(computer_graphics)) yourself to avoid confusion.

closed 2023-05-20 by yuvaltassa

erez-tom, 2023-05-20:
We do some swizzling in our codebase, and the tricky part is always the quaternion (e.g., [here](https://source.corp.google.com/piper///depot/google3/third_party/unity/mujoco/srcs/Runtime/Tools/MjEngineTool.cs;l=154)).  Make sure you do visual sanity-check to your post-swizzling data -- it's easy to get wrong...  Good luck!

areiner222, 2023-05-20:
Thanks for the help!

I ran into the issue with adding a freejoint under a pseudobody that did change of coordinates, so I went with reorienting gravity and adding a rotated camera for the sake of my vis experience. It definitely is a little confusing to keep track of everything.

One issue I am still having is that I'm having trouble with the various rotation representations - in axis-angle representation, everything seems to work as intended but as soon as I convert to quaternions or euler angles I have trouble. I'm expressing some 3dof joints as 3 hinge joints (one per axis) and using euler angles to initialize the orientation but its not initializing as I expect.. if I just use the local axis-angle representations along the bodies of the kinematic tree, however, I get the intended result.

Any ideas what might be happening?

yuvaltassa, 2023-05-20:
That's a bit too vague to be able to help you. If you make a MWE we can try, but I expect that if you make one you'll figure it out yourself 🙂