google-deepmind/mujoco #880: Generating Muscle Activation Data

lalmofee, 2023-05-23:
Hi,

I am a student attempting to analyze muscle activation during an arm movement generated by rotating a crank. Essentially, I am observing the arm muscles in action without actively engaging or activating them, as they naturally move in response to the crank rotation.
I attempted to record the muscle activation during the movement, but I encountered an issue where the resulting array is empty. I would like to know if MuJoCo has the capability to generate muscle activation data.

Thank you

GuillaumeVDu, 2023-05-23:
Hi,
by muscle activation do you mean muscle force?

Vittorio-Caggiano, 2023-05-23:
@lalmofee you can have a look how we do it in MyoSuite: https://github.com/facebookresearch/myosuite/blob/e020c6c8ff9db8f5627cd3561a981fba88438ac8/myosuite/envs/env_base.py#L463 

[MyoSuite](https://github.com/facebookresearch/myosuite) was designed exactly to train musculoskeletal models and might provide some of the functionalities you are looking for

GuillaumeVDu, 2023-05-23:
@lalmofee Muscle activation is an input that you have to give to the model. If you want to generate it check @Vittorio-Caggiano comment.
From what I understand if you want to see the passive effect on crank turning then you will have no muscle activation.

yuvaltassa, 2023-05-24:
I don't understand the OP. You are pulling externally on a passive muscle and expecting to see activation?
This is passive stretching force, the black line in the top left image in [this section](https://mujoco.readthedocs.io/en/latest/modeling.html#muscle-actuators). You would get that for if you moved the arm of a dead person. No activation required or expected.

Am I missing something?

Closing for now.

closed 2023-05-24 by yuvaltassa