google-deepmind/mujoco #882: Running MuJoCo on M1 Mac

dtorre38, 2023-05-23:
Hi,

I'm a student and I'm trying to use MuJoCo for research.

I'm looking for some help with running mujoco on M1 Mac. Unfortunately there are no tutorials on the web.

I've tried modifying the CMakeLists.txt file to run a sample code, but I cannot get it to run. I found one method which works (on v2.2.1), but it is very convoluted. To run the sample code:

1) Place the following in the same folder: libglfw.3.3.dylib, libmujoco.2.2.1.dylib, and a "run" file which is structured as follows:

`reset
make -f makefile clean -C ../sample
make -f makefile -C ../sample
install_name_tool -change @rpath/mujoco.framework/Versions/A/libmujoco.2.2.1.dylib ~/Documents/mujoco-2.2.1/build/mujoco_macosARM/bin/libmujoco.2.2.1.dylib main
install_name_tool -change /opt/homebrew/opt/glfw/lib/libglfw.3.dylib ~/Documents/mujoco-2.2.1/build/mujoco_macosARM/bin/libglfw.3.3.dylib main
./main ../model/sample.xml`

and the makefile is as follows:

`COMMON = -O2 -I../include -L../bin
LDFLAGS = -lglfw.3.3 -lmujoco.2.2.1
CC = gcc
ROOT = -o ../bin/main
CFLAGS = -c -w

all: main_all

main_all: main.o
	$(CC) $(COMMON) $(LDFLAGS) $(ROOT) main.o
main.o: main.c
	$(CC) $(COMMON) $(CFLAGS) main.c
clean:
	rm main.o
`

2) Execute main file in terminal.

This method works, but it's not the most efficient. Has anyone else found an easy way to run MuJoCo on M1 Mac?

dtorre38, 2023-05-23:
The mujoco app works, but that only provides basic functionality. I am referring to running my own models with my own functionality through the terminal.

saran-t, 2023-05-23:
I'm still not clear on what is it that you're asking. You can simulate your own models through the prebuilt binaries by drag-dropping your own XML file.

Are you writing your own C++ program that you want to link against MuJoCo? In that case you could either link against MuJoCo as a framework via `-framework mujoco` (provided that you use `-F` to set the correct path to `mujoco.framework`) or you can just link against the `.dylib` directly using the `-l` flag, as you would any other dynamic library.

If you are a Python programmer the you just need to `pip install mujoco`.

dtorre38, 2023-05-24:
I am programming in C and linking MuJoCo as a framework. This worked fine for my intel Mac, but upon switching to M1 all files are broken. Even a simple case of running a C script linked with MuJoCo will no longer run. The test case is a C script that opens simulate and displays a cube.

To get this to run I have to place the following in the same folder as the run file, C script, and XML:

**libglfw.3.3.dylib
libmujoco.2.2.1.dylib**

The run file contains the following lines (added spaces between each line for readability):

`reset 

make -f makefile clean -C ../sample

make -f makefile -C ../sample

**install_name_tool -change @rpath/mujoco.framework/Versions/A/libmujoco.2.2.1.dylib ~/Documents/mujoco-2.2.1/build/mujoco_macosARM/bin/libmujoco.2.2.1.dylib main

install_name_tool -change /opt/homebrew/opt/glfw/lib/libglfw.3.dylib ~/Documents/mujoco-2.2.1/build/mujoco_macosARM/bin/libglfw.3.3.dylib main** 

./main ../model/sample.xml`

Where the makefile is:

`COMMON = -O2 -I../include -L../bin
LDFLAGS = -lglfw.3.3 -lmujoco.2.2.1
CC = gcc
ROOT = -o ../bin/main
CFLAGS = -c -w

all: main_all

main_all: main.o
(COMMON) (ROOT) main.o
main.o: main.c
(COMMON) $(CFLAGS) main.c
clean:
rm main.o
`

This works, but is different from what is required to run on an Intel Mac. The differences are bolded. I'm sure that there is a simpler way to run mujoco on M1.

saran-t, 2023-05-24:
Can you provide more detail on what you mean when you say it doesn't run? What's the error?

saran-t, 2023-05-25:
Can you try running `cmake` with `-G Ninja`? You might need to also `brew install ninja` first.

dtorre38, 2023-05-25:
I installed ninja and ran as you suggested. Now I'm getting this error:

[image]

dtorre38, 2023-05-31:
> I installed ninja and ran as you suggested. Now I'm getting this error:
> 
> [image]

I found this same error on a stack overflow post: https://stackoverflow.com/questions/73142302/ive-installed-a-library-into-my-usr-local-directory-i-would-now-like-to-inclu

I attempted to implement the suggested fix (below), but it did not work.

`find_package(mujoco REQUIRED)`

dtorre38, 2023-06-01:
This was negligence on my part. I found that I skipped a step: 

Renaming the libmujoco.2.X.X.dylib file to mujoco.dylib.

Unfortunately this still doesn't solve my issue. Now I am obtaining a new error: 

[image]

dtorre38, 2023-06-01:
Due to my negligence I skipped a step when installing MuJoCo: 

Renaming libmujoco.2.X.X.dylib file to mujoco.dylib.

Now the cmake builds successfully, but the executable is not properly created:

[image]

I am unsure on how to proceed.

dtorre38, 2023-06-02:
Alternatively, I attempted to run with a modified version of the Makefile.macos that comes in the sample folder as shown here:

`GLFWROOT ?= $(shell brew --prefix)
MUJOCOPATH ?= /Applications/MuJoCo.app/Contents/Frameworks

CFLAGS := -O2 -F$(MUJOCOPATH) -I$(GLFWROOT)/include -pthread
LDFLAGS := -L$(GLFWROOT)/lib -Wl,-rpath,$(MUJOCOPATH)
LDLIBS := -framework mujoco

all: main

main: main.c
	gcc $(CFLAGS) $(LDFLAGS) $^ $(LDLIBS) -o $@

.PHONY: clean
clean:
	rm -f main`

But I ran into this error:

ld: framework not found mujoco
clang: error: linker command failed with exit code 1 (use -v to see invocation)

This was after completing a fresh install of MuJoCo 2.3.5. Unfortunately, the same error persists, MuJoCo cannot be found/linked.

dtorre38, 2023-06-02:
Completed a fresh install of MuJoCo 2.3.5 and attempted to run with a modified version of the Makefile.macos that comes in the sample folder (shown below). Unfortunately, the same error persists, MuJoCo cannot be found/linked: 

ld: framework not found mujoco
clang: error: linker command failed with exit code 1 (use -v to see invocation)

```GLFWROOT ?= $(shell brew --prefix)
MUJOCOPATH ?= /Applications/MuJoCo.app/Contents/Frameworks

CFLAGS := -O2 -F$(MUJOCOPATH) -I$(GLFWROOT)/include -pthread
LDFLAGS := -L$(GLFWROOT)/lib -Wl,-rpath,$(MUJOCOPATH)
LDLIBS := -framework mujoco

all: main

main: main.c
	gcc $(CFLAGS) $(LDFLAGS) $^ $(LDLIBS) -o $@

.PHONY: clean
clean:
	rm -f main

dtorre38, 2023-06-05:
Found the fix thanks to #313. In the terminal, run the following once:

```
ln -s Versions/Current/libmujoco.2.3.5.dylib /Applications/MuJoCo.app/Contents/Frameworks/mu
joco.framework/mujoco
```

@saran-t mentioned that this issue was resolved with v.2.2.1 in #313, but that wasn't the case for me. I'm not sure how prevalent this issue is, but it might be worth looking into although there is a simple fix thanks to @coolvision.

This is the sample makefile that works:

```
GLFWROOT ?= $(shell brew --prefix)
MUJOCOPATH ?= /Applications/MuJoCo.app/Contents/Frameworks

CFLAGS := -O2 -F$(MUJOCOPATH) -I$(GLFWROOT)/include -pthread
LDFLAGS := -L$(GLFWROOT)/lib -Wl,-rpath,$(MUJOCOPATH)
LDLIBS := -framework mujoco -lglfw

all: main

main: main.c
	gcc $(CFLAGS) $(LDFLAGS) $^ $(LDLIBS) -o $@

.PHONY: clean
clean:
	rm -f main
```

I am still unable to run via cmake. I will post once I get that working as CMakeLists.txt is the preferred method for MuJoCo.

dtorre38, 2023-06-06:
Issue with CMakeLists.txt was due to skipping a step in building from source:

```
cmake --install .
```

Now the mujoco library can be found. This may have also influenced the previous error with the makefile. 

Here is a working CMakeLists.txt file:

```
cmake_minimum_required(VERSION 3.16)
project(template)

# Append the project-specific cmake module path
list(APPEND CMAKE_MODULE_PATH "${PROJECT_SOURCE_DIR}/cmake")

# Specify the link directories for the GSL library
link_directories(${CMAKE_SOURCE_DIR}/../lib/gsl)

# Set the C compiler flags to suppress warnings
set(CMAKE_C_FLAGS -w)

# Specify additional libraries to link against
set(ADD_LIBS -lgsl -lgslcblas -lm)

# Find the Mujoco and GLFW3 packages
find_package(mujoco REQUIRED)
find_package(glfw3 REQUIRED)

# Create the main executable
add_executable(main main.c)

# Set compile options for the main executable
target_compile_options(main PUBLIC ${MUJOCO_SAMPLE_COMPILE_OPTIONS})

# Link libraries to the main executable
target_link_libraries(main ${ADD_LIBS} mujoco::mujoco glfw)
```