google-deepmind/mujoco #883: Unreliable force-torque readings

rollingball-3, 2023-05-25:
Hi,
I'm a student and I'm trying to use MuJoCo for robot simulation.
[image]
Here, a cylinder peg slipping while in contact with the desk(box).

Below is my contact force data read directly from force and torque sensor in mujoco.
[image]
There exists lots of jumping datas which make me feel unreliable.

I have tried to adjust some parameters like **solref** **solimp** **friction** **condim**. But it seems not work pretty well.
How can I get reasonable force readings in this situation ? Thank you for your time.

JeanElsner, 2023-05-25:
In my experience real data looks even more wild.

rollingball-3, 2023-05-26:
@erez-tom Sorry for my late reply. First, cartesian impedance control is used to directly calculate desired joint torque. And I use **motor** actuator to control robot.  The stiffness is set to 100-200. As you said, I have made some attempts on physical franka robots, data was collected from the panda robot itself (estimation) not from the force-torque sensor.
[image]
One thing makes me feel unreasonable is that during silpping , the state changes between contact and non-contact frequently. But it supposed to be contact always(?)

rollingball-3, 2023-05-26:
@yuvaltassa Thank you for your reply. MultiCCD has small influence, while setting `solimp[0]=0` works much better.

closed 2023-05-26 by yuvaltassa