google-deepmind/mujoco #885: Not able to wind the winch properly using rope composite.

haasithp, 2023-05-27:
Hi,

I'm a student and I'm trying to use MuJoCo for Research.
Please find the attached video.
Could you please help me with how to wind the rope and also how to connect the other end to another body using the equality constraint(as seen in the image?

[image]

https://github.com/deepmind/mujoco/assets/99711236/76748e42-9f26-4f8d-a219-91660258d175

quagla, 2023-05-29:
Can you post the XML you are using? Also, the rope is deprecated in favour of the cable, but I have the feeling that you might actually want to use a 1d grid for this application because you actually need to stretch the cable to obtain what you want (i.e. the cable is in tensile stress).

haasithp, 2023-05-29:
So, the rope winded on the winch drum shouldn't have any elasticity. Here, I am trying to simulate something like a slingshot. The rope winds around the drum, stretching the elastic material attached at the other end of the rope, storing enough energy to release an object.

closed 2023-06-16 by haasithp