google-deepmind/mujoco #887: Object being placed on a surface with varying friction underneath

ShuoZheLi, 2023-05-29:
If I have a body such as a box is placed on the top of the plane with two surfaces with different friction connecting, and the object is place in the middle. What will be the friction that the box is experiencing? Consider the box has a very little friction and less than the two surface frictions.
[image]

Balint-H, 2023-06-09:
I'm not from the MuJoCo team but I'll have a go at it. As far as I understand friction is calculated at contact points (see https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-geom). For a box lying flat, 4 contact points are calculated at the touching face's vertices (see image). The effective friction will be mixed based on the touching surfaces' `solmix` parameter. Alternatively if one geom has higher priority then it will determine the contact dynamics.

Assuming the box is aligned with the boundary, and starts completely on surf 1 sliding towards surf 2, it will effectively experience friction 1 only, then the average of friction 1 and 2 as the first 2 vertices cross over, then finally friction 2 only.
[image]

ShuoZheLi, 2023-06-24:
> 

I think your answer make sense to me!

closed 2024-08-30 by kevinzakka