google-deepmind/mujoco #888: How to modify XML files properly?

Mozhdeb2, 2023-05-29:
Hi,

I'm a student and I'm trying to use MuJoCo in an RL project. 

I need to modify some of the mujoco environments. 

I have tried to modify the XML files but the results are not proper.  For example, I want to modify half_cheetah so that its hind legs are smaller than the original half_cheetah. The modified XML file is attached.

The problem I am facing is that when I train this new model using ppo,  the hind legs are not attached properly. A screenshot of the trained agent is given below.  I would appreciate it if you could guide me on how to resolve this issue. Would I need to use solidworks to modify the XML file correctly? 

[image]

saran-t, 2023-06-02:
You need to also modify the `pos` attribute of the `<body>` element containing the geom that you just shortened.

Mozhdeb2, 2023-06-02:
I did change it, and the given picture is what I get after modification. Probably I am not changing the positions correctly. I would appreciate it if you guide me on that.