google-deepmind/mujoco #891: how to set and change the body pos in the code writing while not in the xml?

wukongwong, 2023-05-29:
<body name="trx_hand_body" pos="0.2 0.0 0.2" euler="0 -1.57 0">

I defined the body pos in the xml file as above. Now I wanna change the body pos in Python, the part code is shown as below:

def hand_config_setting(self):
    mujoco.mj_resetData(self.model,self.data)
    self.model.opt.gravity = np.array([0, 0, -9.8])
    #palm
    self.model.body('trx_hand_body').pos = [3.0,0.5,0.8]
    mujoco.mj_step(self.model, self.data)
    mujoco.mj_kinematics(self.model, self.data)
    self.viewer.sync()
    
        # fingers
        for k in range(int(self.policy_timestep / self.sim_timestep)):
            self.do_simulation(self.init_config[6:])
            mujoco.mj_step(self.model, self.data)
        self.viewer.sync()
        return True

However, it seems the hand palm pose in the mujoco I observed does not have any changement after I run this function. Can you pls share what should I do? Thx.

saran-t, 2023-06-02:
Can you please post the XML? I'm finding it difficult to figure out what you're trying to do here.

yuvaltassa, 2023-06-25:
Closing due to lack of activity.

As @saran-t says, not clear what you are trying to do, but I think you might be wanting to use [a mocap body](https://mujoco.readthedocs.io/en/latest/modeling.html?highlight=mocap#mocap-bodies)

closed 2023-06-25 by yuvaltassa