google-deepmind/mujoco #894: Mujoco Viewer black screen

nsk126, 2023-06-02:
Mujoco viewer seems to be bugged. I've been trying to load my xml files into it but it seems to open a black screen? Even the command `python -m mujoco.viewer` seems to open a window like the following:

[image]

The result is the same when i load an xml file like `walker2d.xml` like this:

```python
import mujoco
import mujoco.viewer as vi

model = mujoco.MjModel.from_xml_path('walker2d.xml')
vi.launch(model)
```

kevinzakka, 2023-06-02:
Do you have an AMD cpu? Can you launch again with [this](https://github.com/deepmind/mujoco/issues/639#issuecomment-1514622779)?

nsk126, 2023-06-02:
Yes, I'm on a Ryzen 9, also Radeon GPU. But the same script worked a couple weeks ago. Not sure what changed now.

I'm not sure what you mean by launch with? Do you mean to say to set env variables `MUJOCO_GL_DEBUG=1` and `LIBGL_DEBUG=verbose`?

saran-t, 2023-06-02:
Can you launch a command prompt, set those environment variables, run simulate in the same terminal, and see if you get any diagnostic printouts?

nsk126, 2023-06-02:
I don't get any print outputs.

BenZen-3, 2023-06-06:
I have an AMD cpu and ran into the same issues. Managed to fix it by forcing simulator to use my Nvidia gpu. If you're on Windows you can do it through the graphics settings.

YusseffRuiz, 2023-08-08:
Just to help with this problem.
I used Mujoco inside PyCharm with Anaconda in my Windows OS. I had the same black screen, using myosuite as well.
I solved it, jumping into NVidia Control Pannel >>> Control 3D configuration >> Program configuration >>
added Python on the desired environment and Pycharm to NVIDIA processor.

It worked great afterwards!

Good luck
[image]

nsk126, 2023-08-08:
An alternative i found was https://github.com/rohanpsingh/mujoco-python-viewer. Seems to work just fine somehow.

saran-t, 2023-10-20:
Let's track progress in #639, I've asked for a bit of help debugging over there. I'm reasonably hopeful that we'll finally get this fixed!

closed 2023-10-20 by saran-t