google-deepmind/mujoco #895: What is quat2vel mean?

Uptomylimit, 2023-06-02:
Hi,

I'm a student and I'm trying to use MuJoCo for inverse kinematics.

I'm looking for some help with the Mujoco function mju_quat2vel.

The inverse_kinematics.py of mujoco use mju_quat2vel to calculate the rot error for inverse kinematics method based on jacobian.

I find that the result of mju_quat2vel doesn't equal to euler angle, or the axis of the axis-angle, or fixed axis angle.

Can you show me how you calculate the orientation error or some papers your code based on?

saran-t, 2023-06-02:
This function is essentially the logarithm map. I think the easiest way to understand it is to read its source code in conjunction with `mju_subQuat` [here](https://github.com/deepmind/mujoco/blob/main/src/engine/engine_util_spatial.c#L138)

closed 2023-06-06 by saran-t

Uptomylimit, 2023-06-09:
Thanks for your clear answer! The code is very clear!