google-deepmind/mujoco #897: C++ threading performance is no better than Python

PeterMitrano, 2023-06-02:
I'm a PhD student using mujoco for deformable object manipulation. I've noticed the documentation says (and @quagla  told me directly) that doing multiple rollouts in parallel in C++ should be faster than in python, but my testing shows them to be comparable or even slightly faster in python. I assume I am doing something wrong, but I describe my methods below and hopefully someone can explain what I'm seeing:

Using the `humanoid100.xml` file I can run the C++ speed test like this on my 8 core T15 laptop:

```
pmitrano@pmitrano:~/projects/mujoco/build (main)$ ./bin/testspeed model/humanoid100.xml 10000 8

Running 10000 steps per thread at dt = 0.005 ...

Summary for all 8 threads

 Total simulation time  : 12.83 s
 Total steps per second : 6235
 Total realtime factor  : 31.17 x
 Total time per step    : 0.1604 ms

Details for thread 0

 Simulation time      : 12.47 s
 Steps per second     : 802
 Realtime factor      : 4.01 x
 Time per step        : 1.2473 ms
```

Then in python, something which I _think_ is comparable (this is probably where the issue arises...)

```
#!/usr/bin/env python3
import argparse
import concurrent
import copy
from concurrent.futures.thread import ThreadPoolExecutor
from time import perf_counter

import mujoco
import numpy as np

N_TIME = 10000
N_ACTUATORS = 21

def rollout_one_trajectory(model, data, controls):
    qs = []
    for t in range(N_TIME):
        control_t = controls[t]
        np.copyto(data.ctrl, control_t)
        mujoco.mj_step(model, data)
        qs.append(data.qpos.copy())
    return qs

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("xml_path", type=str)

    args_lists = parser.parse_args()

    model = mujoco.MjModel.from_xml_path(args_lists.xml_path)
    data0 = mujoco.MjData(model)

    nthreads = 8
    controls = np.random.randn(N_TIME, N_ACTUATORS)

    def call_rollout(ctrl):
        data = copy.copy(data0)
        rollout_one_trajectory(model, data, ctrl)

    args_lists = [(controls,)] * nthreads

    print(f"Running {N_TIME} steps per thread at dt = {model.opt.timestep} ...")

    t0 = perf_counter()
    with ThreadPoolExecutor(max_workers=8) as pool:
        futures = []
        for args_list in args_lists:
            futures.append(pool.submit(call_rollout, *args_list))
        for future in concurrent.futures.as_completed(futures):
            future.result()
    total_dt = perf_counter() - t0
    steps_per_second = N_TIME * nthreads / total_dt
    rtf = steps_per_second * model.opt.timestep
    time_per_step = total_dt / (N_TIME * nthreads) * 1000

    print(f"Total simulation time  : {total_dt:.2f} s")
    print(f"Total steps per second : {steps_per_second:.0f}")
    print(f"Total realtime factor  : {rtf:.3f} x")
    print(f"Total time per step    : {time_per_step:.3f} ms")

if __name__ == "__main__":
    main()
```
Run this script like `python ./perf_rollout.py humanoid100.xml`
And it's quite fast! 

```
Running 10000 steps per thread at dt = 0.005 ...
Total simulation time  : 8.15 s
Total steps per second : 9818
Total realtime factor  : 49.091 x
Total time per step    : 0.102 ms
```

I'd appreciate any insides into why there's no performance benefit in C++, or why my comparison is not valid. Thanks!

saran-t, 2023-06-02:
Are you building from source or running our rebuilt binaries? (The question applies to both the C++ and Python tests)

Just to eliminate possible compilation differences, can you rerun these tests using our prebuilt binaries both in C++ and Python?

PeterMitrano, 2023-06-02:
I re-tested with 2.3.5 and results are slightly different, though based on the variance between runs I am not confident there is any significant change.

C++
```
pmitrano@pmitrano:~/Downloads/mujoco-2.3.5/bin$ ./testspeed humanoid100.xml 10000 8

Running 10000 steps per thread at dt = 0.005 ...

Summary for all 8 threads

 Total simulation time  : 8.40 s
 Total steps per second : 9519
 Total realtime factor  : 47.60 x
 Total time per step    : 0.1051 ms
```

Python
```
Running 10000 steps per thread at dt = 0.005 ...
Total simulation time  : 10.49 s
Total steps per second : 7628
Total realtime factor  : 38.140 x
Total time per step    : 0.131 ms
```

saran-t, 2023-06-02:
Going from 12.8s to 8.4s (comparing your original C++ numbers with the prebuilt numbers) is pretty significant IMO. Can you run each one a few times in a row to see what kind of variability you're getting? (Not super scientific, but at least we'll eliminate e.g. effects of disk caching).

yuvaltassa, 2023-06-02:
Hi @PeterMitrano,

Since you are doing open loop rollouts, you could also take a look at `rollout` ([docs](https://mujoco.readthedocs.io/en/latest/python.html#open-loop-rollouts), [test](https://github.com/deepmind/mujoco/blob/main/python/mujoco/rollout_test.py).

Note there is an explicit test for multithreading in the test file.

That might also be a relevant data point.

PeterMitrano, 2023-06-02:
Here's an attempt to compare *10 samples* of the timing between the now three options. Showing average for the first 4 stats then standard deviation at the end.

Python: my own loop
```
Average Total simulation time  : 10.90 s
Total steps per second : 7339
Total realtime factor  : 36.697 x
Total time per step    : 0.136 ms
Std Total sim time     : 0.706 ms
```

Python: using `rollout`
```
Average Total simulation time  : 10.31 s
Total steps per second : 7761
Total realtime factor  : 38.805 x
Total time per step    : 0.129 ms
Std Total sim time     : 0.913 ms
```

C++
```
Average Total simulation time  : 10.31 s
Total steps per second : 7762
Total realtime factor  : 38.812 x
Total time per step    : 0.129 ms
Std Total sim time     : 0.291 ms
```

The different does not seem significant to me.

yuvaltassa, 2023-06-02:
Taken at face value this just seems to imply that Python threading is not so bad. Better than what I expected at least 🙂

I'm guessing that you'd see a bigger difference with a smaller model. Can you rerun with regular humanoid.xml? For that model I also have a better sense of what timing tends to look like more generally.

PeterMitrano, 2023-06-02:
Here are the results for normal `humanoid.xml`

Python custom rollout:
```
Average Total simulation time  : 1.20 s
Total steps per second : 66479
Total realtime factor  : 332.394 x
Total time per step    : 0.015 ms
Std Total sim time     : 0.036 ms
```

Python default rollout:
```
Average Total simulation time  : 0.38 s
Total steps per second : 208171
Total realtime factor  : 1040.856 x
Total time per step    : 0.005 ms
Std Total sim time     : 0.015 ms
```

C++
```
Average Total simulation time  : 0.45 s
Total steps per second : 178571
Total realtime factor  : 892.857 x
Total time per step    : 0.006 ms
Std Total sim time     : 0.015 ms
```

This suggests that my custom rollout isn't as fast in this case, but the threading itself seems comparable!
I suppose when the time per step is small, the overhead of python matters more.

We can consider this closed, since I've satisfied my curiosity, but hopefully this investigation is useful to others as well.

yuvaltassa, 2023-06-02:
Thanks @PeterMitrano, much appreciated.

One day we'll get around to making a proper benchmarks page. The tricky bit is that to make it useful it would need to be both multiplatform (obvious) auto-generated (since we are constantly making speed improvements) and contain all the data from previous versions (so speed improvements (and regressions, if any) are visible across versions). The combination of these desiderata makes this a non-trivial bit of work...

closed 2023-06-02 by yuvaltassa