google-deepmind/mujoco #898: Add lights to a scene

v-r-a, 2023-06-02:
Hi,

I'm a student using MuJoCo in C and Python.

I'm looking for some help with adding lights to a scene. I'm trying to render good images of a robot in Python.

MuJoCo in Python:
I am aware that we can add lights in the XML file. However, I would like to add them through a code the way we can define a camera using mjvCamera.

I tried defining an instance of mjvLight, added one more light to the scene, and defined its properties like pos, dir, etc.
light_key = mujoco.MjvLight();
renderer.scene.nlight +=1

However, the following throws an error that this object does not support item assignment.
renderer.scene.lights[1] = light_key

MuJoCo in C:
I tried similar steps in vain. Scrolled source code.
Also realised there is no mj_defaultLight()
Also checked out mj_makeLights() and couldn't understand how one can add extra lights by themselves.

Requesting a minimal working example in Python if such a feature is supported.
If it is not supported, can we define lights in XML and then change their properties through the code?

Thank you

yuvaltassa, 2023-06-02:
I've never tried _adding_ lights programmatically, but putting a bunch of lights in XML and modifying at runtime in the mjModel should just work.

Let us know if you hit a snag.

I'll also look into actually adding lights, but since there is no cost to the physics, I'd think defining them in advance and modifying is a very reasonable solution.

v-r-a, 2023-06-02:
Attaching a Python example (Jupyter Notebook) for tweaking various values. Using the latest humanoid.xml.

I could observe the change in `light_diffuse`. I could not observe the change in any other properties of a given light.

Some explanation on the '// lights'  section in mjModel could help.

I guess modifying mjModel in runtime is working. Not sure.

Any help is appreciated.

Thank you

[Image_export_humanoid.zip](https://github.com/deepmind/mujoco/files/11639272/Image_export_humanoid.zip)

v-r-a, 2023-06-03:
Light adjustments are working as expected. One needs to know which values in the '// lights' section in mjModel are computed automatically based on the type of light defined.

I've attached the updated example.

[image_export_example_python.zip](https://github.com/deepmind/mujoco/files/11644270/image_export_example_python.zip)

The possibility of adding lights programmatically is yet to be confirmed.

closed 2023-06-03 by v-r-a